espressif/idf-eclipse-plugin #62: Build not configured correctly in eclipse Windows 10

aarizaq, 2020-03-18:
I cannot build a basic project in windows 10 with eclipse, in linux it works for me.

I can build the projects from the command line

Installation Steps.
1.- I have clone esp-idf from github
2.- I have downloaded and install esp-idf-tools-setup-2.3.exe.
3.- Plugin Install version   Espressif IDF Plugins for Eclipse	1.0.0.202003110736	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD
4.- Execute Help > Espressif IDF Tools Manager > Install Tools
5.- Execute Help >  Espressif IDF Tools Manager > List Installed Tools. Output
Executing D:\Users\aariz\AppData\Local\Programs\Python\Python37\python.exe D:\SysGCC\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

Start Eclipse.
1.- start esp-idf command prompt. 
2.- >eclipse.exe

[eclipseConfiguration.txt](https://github.com/espressif/idf-eclipse-plugin/files/4348746/eclipseConfiguration.txt)

aarizaq, 2020-03-18:
I

aarizaq, 2020-03-18:
Finally I have reinstall all and now it works, the only think that doesn't work correctly is the configuration editor in eclipse,

closed 2020-03-18 by aarizaq

isoftproject, 2020-09-22:
This behaviour keeps even though you reinstall the Eclipse

isoftproject, 2020-09-22:
I solved this issue setting the User Defined Toochains following the steps (Configuring Core Build Toolchains) in:https://github.com/espressif/idf-eclipse-plugin#cmakeproject

isoftproject, 2020-09-22:
For any reason I don't have cofigured that (User Defined Toolchains)

wojciech-angelo, 2021-01-13:
it doesn't work perfectly... ;/

hzlijianjun, 2022-07-30:
Even reinstalling with Eclipse C/C++2022-06, The issue keeps on.
Finally I found my problem is that I has not setted launch project. 
 
[image]

Lsitar, 2022-12-21:
Hi, I also had this problem "Build not configured correctly".
Espressif-IDE 2.7.0 (Build id: 20221110-1527) on Windows 10.
I re-installed whole ESP-IDF, but old workspace with projects remained, also the launch configuration and paths were preserved.
For me worked hints from issue #119:
```
To reset the configuration, please do the following.
1. Go to Eclipse preferences
2. C/C++ Build > Environment
3. Delete all variables configured here
4. Run this: Help > ESP-IDF Tools Manager > Install tools (this will reconfigure environment variables)
5. Run the project again by selecting the esp target
```