espressif/idf-eclipse-plugin #63: ESP IDF Eclipse plugin v1.0.0 not working on Eclipse 2020 03 / Fedora 11

dolhastelian, 2020-03-21:
**Describe the bug**
ESP IDF Eclipse plugin v1.0.0 not working in Eclipse 2020-03

**To Reproduce**
Steps to reproduce the behavior:
0. Test system: Fedora 31
1. Install Eclipse
Eclipse IDE for C/C++ Developers

Version: 2020-03 (4.15.0)
Build id: 20200313-1211

2. Install ESP-IDF Eclipse plugin v1.0.0
3. install git, esp-idf 4.0 and python 3.7
4. Open Install tools dialogue, select esp-idf git and python
5. Press Install Tools
6. - nothing happens, no task started, no error

**Expected behavior**
install.sh should be triggered as a task in Eclipse

jwestmoreland, 2020-03-22:
Hello,

I've noticed a few things that I think are worth mentioning here under Win10:

1)  If I have all of the tools installed and then install the IDF - I see something similar as what is reported here - nothing seems to happen.  Also, when doing New->... - the ESP IDF Project template never shows up.  It's like the plugin halts right after it determines all of the tools are the.

2)  After installation - doing a Build All sometimes results in the target not set error even though the target has been set.  But, if you press the little hammer in the top left corner, the build proceed and the target is found ok.  

Regards,
John W.

kolipakakondal, 2020-03-23:
Hello @jwestmoreland 
Thanks for reporting the issue.

1. Could you try to reset the perspective after installing the plugins as mentioned here to see the new project menu option? https://github.com/espressif/idf-eclipse-plugin#create-a-new-project

> Window > Perspective > Reset Perspective..

2.  I'll verify the "Build All" option.

As you mentioned, "install tools" is not triggering any action, could you please check the error log to see anything reported.
https://github.com/espressif/idf-eclipse-plugin#error-log

JAICHANGPARK, 2020-03-24:
I have same problem about this issue
- Eclipse 2020-03
- Plugin 1.0.0 

1. First,  Install Tools nothing happens, no task started, no error
2. Second, invisible next button when I create new IDF Project 

[image]

Thank you

jwestmoreland, 2020-03-24:
Hello All,

I've seen #2 above as well @JAICHANGPARK .

But, if you go to New->Project... and follow the new project wizard, the ESP IDF Project is there and that seems to work even though the last time I tried that when I build it fails to find the ninja installation.  cmake claims there's no ninja in the path even though there is.

And, @kolipakakondal , even though I thought I did that - I just did it (again maybe?) and it did work - I can see the ESP IDF Project listing now where I couldn't before.

Thanks,
John W.

kolipakakondal, 2020-03-24:
> I have same problem about this issue
> 
> * Eclipse 2020-03
> * Plugin 1.0.0
> 
> 1. First,  Install Tools nothing happens, no task started, no error
> 2. Second, invisible next button when I create new IDF Project
> 
> [image]
> 
> Thank you

This can only happen when IDF_PATH not defined in the CDT build environment variables. And this is configured as part of the tools install from eclipse.

kolipakakondal, 2020-03-24:
@dolhastelian I couldn't reproduce with 2020-03 version on macOS. Please provide the error log.

dolhastelian, 2020-03-25:
Hello,
I had a problem with the path for python, after checking the error log.
I manually selected the correct path and it is now working.

One note: by default, the dialogue box will pre-fill git, idf and python with a predefined path.
If the plugin is not actually searching for the right path to prefill the fields, it would be better to leave the fields empty so that the user is forced to manually select the right path for the tools.
Please share your opinion, I would recommend to close the ticket and, up to your decision, a new one could be opened to fix the note above.

Thank you for your great support and quick reply.

jwestmoreland, 2020-03-25:
To All:

Here's something I noticed on one of my machines but I was unable to get a screen capture.

When Build All was clicked - it picked the $IDF_PATH path.
When the Hammer is clicked to kick off the build (eclipse) - the WorkSpace Path - I think that's
$(Workspace_Loc) was picked.  (Or vice-versa...)

I think this could be a source of some of the install issues.  Even though I've made sure IDF_PATH, IDF_TOOLS_PATH, etc, are all in the environment before installation - I think the above can still happen.

Hope this helps in resolving this issue.

Regards,
John W.

DanHajek, 2020-03-25:
To All:
A have same problem. It look like there is some problem between "config dialog" and IDF_PATH variable assigment problem.
Workaround:
Close all projects.
go to: Window -> Preferences -> C/C++ Build -> Enviroment
Add new variable IDF_PATH with path to your IDF install.
Check "Replace native env..." (not sure if this is required)
Now go to plugin install tools dialog and click install

Have a nice day
Dan

jwestmoreland, 2020-03-26:
@DanHajek,

I checked this on the laptop I'm seeing this problem - that ENV var was already set correctly and all of the symptoms here still occur.  

The install still apparently doesn't do anything that can be visually verified.

Regards,
John W.

kolipakakondal, 2020-03-26:
Guys, whoever is facing the install tools issue, please provide the error log. https://github.com/espressif/idf-eclipse-plugin#error-log

jwestmoreland, 2020-03-26:
@kolipakakondal,

I've attached the error log from the laptop I'm seeing this on.
I tried doing a new install from scratch.

[workspace_error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/4385304/workspace_error_log.log)

Regards,
John W.

kolipakakondal, 2020-03-26:
Thanks @jwestmoreland for attaching the log. I'll check and let you know.

jwestmoreland, 2020-03-26:
Sure @kolipakakondal - thanks for taking a look.

RBasile, 2020-03-28:
Hi,
I don't know if this can help but for me, the problem of the installation doing nothing come from the fact the git location was already filled with the good path 
so for me, it was good
but it's was not took into account and as soon I re-enter the path it works normally.

the differance before and after I re-enter the path.
[image]
[image]

jwestmoreland, 2020-03-29:
@RBasile,

I tried that - and you're right - it helped and the install script did run.

Regards,
John W,

kolipakakondal, 2020-03-30:
> Hi,
> I don't know if this can help but for me, the problem of the installation doing nothing come from the fact the git location was already filled with the good path
> so for me, it was good
> but it's [...]

There is confusion around the git path. What you see is just a suggestion in the text path and that's not a real path. We will remove and avoid this confusion going forward. Thanks for raising this.

kolipakakondal, 2020-03-30:
@jwestmoreland Sorry for the confusion. I've looked into the error log you've provided. It's the problem with the git path. Please make sure you provide the correct path.

kolipakakondal, 2020-03-31:
We have released v1.0.1 with some fixes related to this bug. Please check and update.https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.0.1

closed 2020-03-31 by kolipakakondal