espressif/idf-eclipse-plugin #64: IDF under Win10 - Can't Include libesp32_at_core.a (ESP-AT)

jwestmoreland, 2020-03-22:
Hello All,

Eclipse IDE for C/C++ Developers

Version: 2020-03 (4.15.0)
Build id: 20200313-1211

ESP-IDF v4.0 (rel)
ESP-AT v2.0.0-esp32
Plugin:  v1.0.0-rc2

1.  Install Eclipse.
2.  Follow all of the instructions to install the ESP-IDF plugin.
3.  Build the hello_world example and run successfully on the target.
4.  Try one of the examples in the ESP-AT environment.
5.  Build fails with not including the main link lib for the ESP-AT.

After setting the build target so all of the ESP-IDF includes, libs (built), etc. are done - no matter what I do the main lib file for the ESP-AT environment is never added.  I've been able to change the ninja.build file - but in this case more errors are generated after I do that.  That does work from just running from the command line - I've made s good build that way (IDF v3.3.1 and v2.0 of the AT).

It would appear the ESP-IDF plugin isn't aware of the ESP-AT target.

In the IDF plugin - it would be nice to expose all of the include file paths plus a way to add a lib without going through the convoluted process of the cmakelists.txt syntax which pretty much doesn't work for a lib in which you don't have source files.

Thanks,
John W.

igrr, 2020-03-22:
Hi John,

Please refer to this example for adding a pre-built library to your CMake
project:
https://github.com/espressif/esp-idf/tree/release/v4.0/examples/build_system/cmake/import_prebuilt

jwestmoreland, 2020-03-22:
Hello igrr,

First - thanks for the fast response.  Wow, a google search on that didn't work for me...

Has someone posted a process for the ESP-AT for this?  I tried using the cmake editor but obviously didn't press p to bring up prebuilt.

I'll follow the example and report back.

Thanks!
John W.

jwestmoreland, 2020-03-23:
Hello igrr,

OK - that's done - and now I'm getting errors like this:

c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.hid_dev_send_report+0x8): undefined reference to `esp_ble_gatts_send_indicate'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_at_hidd_event_callback+0x28): undefined reference to `esp_ble_gap_set_device_name'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_at_hidd_event_callback+0x2c): undefined reference to `esp_ble_gap_config_adv_data'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_at_hidd_event_callback+0x30): undefined reference to `esp_ble_gap_start_advertising'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_at_hid_gap_event_handler+0x1c): undefined reference to `esp_ble_gap_security_rsp'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.hidd_register_cb+0x4): undefined reference to `esp_ble_gatts_register_callback'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_register_callbacks+0x8): undefined reference to `esp_ble_gatts_app_register'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.hidd_le_create_service+0x4): undefined reference to `esp_ble_gatts_create_attr_tab'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_prf_cb_hdl+0x1c): undefined reference to `esp_ble_gap_config_local_icon'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_prf_cb_hdl+0x20): undefined reference to `esp_ble_set_encryption'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_prf_cb_hdl+0x24): undefined reference to `esp_ble_gatts_start_service'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_profile_deinit+0x0): undefined reference to `esp_ble_gatts_stop_service'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_profile_deinit+0x4): undefined reference to `esp_ble_gatts_delete_service'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.esp_hidd_profile_deinit+0x8): undefined reference to `esp_ble_gatts_app_unregister'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.at_setupBleHidInit+0x10): undefined reference to `esp_bt_controller_init'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.at_setupBleHidInit+0x14): undefined reference to `esp_bt_controller_enable'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.at_setupBleHidInit+0x18): undefined reference to `esp_bluedroid_init'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.at_setupBleHidInit+0x1c): undefined reference to `esp_bluedroid_enable'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/libesp32_at_core.a(at_blehidCmd.o):(.literal.at_setupBleHidInit+0x20): undefined reference to `esp_ble_gap_register_callback'
[... cut ...] at_setupCmdCipserver':
(.text.at_setupCmdCipserver+0x113): undefined reference to `lwip_setsockopt_r'
c:/esptools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/
....
says 241 before it stops.

Which is interesting since I thought everything else was there.

At least this is progress.

Also - the add_prebuilt... doesn't show up in the 4.0 cmake editor by default which explains why I didn't find it.  

Thanks for pointing that out.

Regards,
John W.

jwestmoreland, 2020-03-23:
Hello All,

For inquiring minds that want to know - this is what ended up working:

add_prebuilt_library(prebuilt "c:/esp-tot/esp-at/components/at/lib/libesp32_at_core.a"
             PRIV_REQUIRES esp32 esp_wifi esp_https_server esp_eth mdns lwip fatfs openssl esp_https_ota )

At least it built.  Haven't debugged yet.

I went to the cmake.org site and searched for add_prebuilt_library from their main search thingy and it turned up NADA.

Regards,
John W.

jwestmoreland, 2020-03-25:
Hello All,

add_prebuilt_library

Solved this issue - so I think it's OK to close this thread now.

Thanks!
John W.

closed 2020-03-25 by jwestmoreland