espressif/idf-eclipse-plugin #67: Can't install link get 403 error

manobendro, 2020-03-31:
Can't install plugin in eclipse. the link "https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/" doesn't work

igrr, 2020-03-31:
Hi @manobendro, please input the link into Eclipse as shown here: https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#installing-idf-plugins-using-update-site-url

The link won't open in the browser, but it does work in Eclipse.

manobendro, 2020-04-05:
I got this error in eclips 
" Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target "

kolipakakondal, 2020-04-06:
Hello @manobendro 

This would have caused by the Java version or java cert issues. https://esp32.com/viewtopic.php?f=13&t=12327&start=10#p50137

https://stackoverflow.com/questions/6908948/java-sun-security-provider-certpath-suncertpathbuilderexception-unable-to-find

Let me know what's the java version you're using?

manobendro, 2020-04-10:
> Hello @manobendro
>
> This would have caused by the Java version or java cert issues. https://esp32.com/viewtopic.php?f=13&t=12327&start=10#p50137 [...]

java version "1.8.0_77"
Java(TM) SE Runtime Environment (build 1.8.0_77-b03)
Java HotSpot(TM) 64-Bit Server VM (build 25.77-b03, mixed mode)

kolipakakondal, 2020-04-10:
That's a pretty old Java 8 version. Please update to the latest Java 8 Update 241 (8u241)
https://java.com/en/download/faq/release_changes.xml

FYI, The plugin also works with any latest Java version - Java 9, 10 and 11

manobendro, 2020-04-10:
Problem fixed by upgrading Java SE(11 LTS)

kolipakakondal, 2020-04-13:
Awesome! All the best

closed 2020-04-13 by kolipakakondal

ktgrhsd, 2020-04-26:
Facing similar problem like manobendro.

Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

My java version,

java version "1.8.0_251"
Java(TM) SE Runtime Environment (build 1.8.0_251-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.251-b08, mixed mode)