espressif/idf-eclipse-plugin #68: Show Build size output

versamodule, 2020-04-01:
The typical build command line is "idf.py build" But I want to use the option size. 
"idf.py build size"

Where can I add the option "size" when clicking the build icon ?

Thanks

kolipakakondal, 2020-04-03:
Hi @versamodule 

The current plugin doesn't have this feature but you can expect this in the next release.
Thanks for raising this.

Thanks

TobiasBoeschMiele, 2020-04-23:
(On Windows 10):
When I
* installed the tools using the [ESP-IDF Tools Installer](https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/windows-setup.html#esp-idf-tools-installer) and then
* install the [eclipse plugin](https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugins-using-update-site-url) and then
* [install the IDF tools through the eclipse plugin](https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf-tools)

the plugin found/used the existing installation.
This setup makes me able to use the IDF command prompt from the start menu. There one can ask for the app size: `idf.py size`.
So I can use eclipse for convenience and the normal tools for completeness.

kolipakakondal, 2020-06-03:
Hello, We have released v1.1.0,  this includes the "Application size analysis" editor to understand the application memory footprint along with the other features.

This also added with build configuration tab where it allows us to provide custom CMake arguments.
Check this https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal