espressif/idf-eclipse-plugin #69: Can not finalize install of ESP IDF Tools

cococococococo, 2020-04-05:
When I attempt to install ESP IDF Tools, in the console it stay "bricked" at this line and doesn't seems to finish the install:

Executing C:\Users\user\AppData\Local\Programs\Python\Python37\python.exe C:\esp-idf\tools\idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=C:\Users\user\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200309\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE=1
IDF_PYTHON_ENV_PATH=C:\Users\user\.espressif\python_env\idf4.2_py3.7_env
PATH=C:\Users\user\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin;C:\Users\user\.espressif\tools\xtensa-esp32s2-elf\esp-2020r1-8.2.0\xtensa-esp32s2-elf\bin;C:\Users\user\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\user\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\user\.espressif\tools\cmake\3.16.4\bin;C:\Users\user\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200309\openocd-esp32\bin;C:\Users\user\.espressif\tools\ninja\1.10.0\;C:\Users\user\.espressif\tools\idf-exe\1.0.1\;C:\Users\user\.espressif\tools\ccache\3.7\;C:\Users\user\.espressif\python_env\idf4.2_py3.7_env\Scripts;C:\esp-idf\tools;%PATH%

kolipakakondal, 2020-04-06:
Hello,

As far as your issue concerned - the install process is already completed and successful. The output of the command **idf_tools.py export --format=key-value** is shown on the console.

Install process status represented by the **Console** **Terminate** button.
red - In process
grey - terminated

However, I would like to take your feedback to represent this more intuitive fashion. Probably, by showing the text message in the console after the job completed.

cococococococo, 2020-04-06:
Well a "Configuration" or "Install " "Done" would be best espcially as  in the Readme you mention that $PATH should be replaced... this part is not very clear.
Also your username Kondal is in the readme you better replace it by a variable and mention...otherwise some newbies would copy paste those path without thinking..

kolipakakondal, 2020-04-06:
>  in the Readme you mention that $PATH should be replaced... this part is not very clear.

Sure, I will review the documentation on this if it's not very clear. $PATH replacement task is done as part of the install tools process by the plugin itself. Users needn't make any modifications here unless it's required.

I'll also check the username in the doc

Thank you.

kolipakakondal, 2020-06-03:
We have improved the tools console messages with v1.1.0 release.

Check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal