espressif/idf-eclipse-plugin #70: Wrong Ninja Path

axbusch, 2020-04-06:
ok, i updated win 7 to win10

i reinstalled eclipse and the idf tools. here is the output of list tools
```
Executing C:\python3_7\python.exe D:\esp\idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
```
The folder when i install the tools via Eclipse is
C:\Users\funky\.espressif\tools\

When i want to build my project

>  Building in: D:\esp\remote\esp32_mp3_decoder\build
> cmake --build . -- -v
> Das System kann die angegebene Datei nicht finden
> CMake Error: Generator: execution of make failed. Make command was: "D:/esp/toolchain/.espressif/tools/ninja/1.9.0/ninja.exe" "-v"

"D:/esp/toolchain/.espressif/tools/ninja/1.9.0/" was the old path on the win7 installation but i don't know where this path is still set. where is this path taken from?

kolipakakondal, 2020-04-07:
Hi @axbusch 
Could you check the PATH from the environment variables?  This might be still having old and new paths. 
`Preferences > C/C++ Build > Environment > PATH`

axbusch, 2020-04-07:
there was the old path. i removed all the variables, reinstalled the tools but still the same behavior

kolipakakondal, 2020-04-08:
@axbusch After removing the PATH manually and reinstalling the tools from eclipse - does the PATH still have the old entries?  Could you please provide the complete PATH value?

leenowell, 2021-01-30:
Hi,

I had exactly this problem and manually fixing it or indeed removing it and reinstalling the tools did not help. I ended up deleting the build folder and rebuilding and it worked fine.  Before deleting the build folder I did a grep for all files in my project with ninja in to try and identify the root cause so hopefully this will help find the bug?

```
4$ grep -r ninja .

binary file ./build/.ninja_deps matches
./build/.ninja_log:# ninja log v5
./build/.ninja_log:74	25774	1611861443674457600	build.ninja	2e2d2707f897cc00
Binary file ./build/bootloader/.ninja_deps matches
./build/bootloader/.ninja_log:# ninja log v5
./build/bootloader/.ninja_log:12	4942	1610485046308597200	build.ninja	70643785a1969473
./build/bootloader/.ninja_log:75	13919	1611435647524265700	build.ninja	70643785a1969473
./build/bootloader/.ninja_log:7	5681	1611861506802916600	build.ninja	70643785a1969473
./build/bootloader/build.ninja:ninja_required_version = 1.5
./build/bootloader/build.ninja:include rules.ninja
./build/bootloader/build.ninja:build build.ninja: RERUN_CMAKE | /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeASMInformation.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeCInformation.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeCXXInformation.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeCommonLanguageInclude.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeGenericSystem.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeInitializeConfigs.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeLanguageInformation.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeSystemSpecificInformation.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeSystemSpecificInitialize.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Compiler/CMakeCommonCompilerMacros.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Compiler/GNU-ASM.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Compiler/GNU-C.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Compiler/GNU-CXX.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Compiler/GNU.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/ExternalProject.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/FindGit.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/FindPackageMessage.cmake /home/lee/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/Platform/Generic.cmake /home/lee/esp/esp-idf/.git/HEAD /home/lee/esp/esp-idf/.git/modules/components/asio/asio/HEAD /home/lee/esp/esp-idf/.git/modules/components/bootloader/subproject/components/micro-ecc/micro-ecc/HEAD /home/lee/esp/esp-idf/.git/modules/components/bt/controller/lib/HEAD /home/lee/esp/esp-idf/.git/modules/components/bt/host/nimble/nimble/HEAD /home/lee/esp/esp-idf/.git/modules/components/cbor/tinycbor/HEAD /home/lee/esp/esp-idf/.git/modules/components/coap/libcoap/HEAD /home/lee/esp/esp-idf/.git/modules/components/esp_wifi/lib/HEAD /home/lee/esp/esp-idf/.git/modules/components/esptool_py/esptool/HEAD /home/lee/esp/esp-idf/.git/modules/components/expat/expat/HEAD /home/lee/esp/esp-idf/.git/modules/components/json/cJSON/HEAD /home/lee/esp/esp-idf/.git/modules/components/libsodium/libsodium/HEAD /home/lee/esp/esp-idf/.git/modules/components/lwip/lwip/HEAD /home/lee/esp/esp-idf/.git/modules/components/mbedtls/mbedtls/HEAD /home/lee/esp/esp-idf/.git/modules/components/mqtt/esp-mqtt/HEAD /home/lee/esp/esp-idf/.git/modules/components/nghttp/nghttp2/HEAD /home/lee/esp/esp-idf/.git/modules/components/protobuf-c/protobuf-c/HEAD /home/lee/esp/esp-idf/.git/modules/components/spiffs/spiffs/HEAD /home/lee/esp/esp-idf/.git/modules/components/tinyusb/tinyusb/HEAD /home/lee/esp/esp-idf/.git/modules/components/unity/unity/HEAD /home/lee/esp/esp-idf/.git/modules/examples/build_system/cmake/import_lib/main/lib/tinyxml2/HEAD /home/lee/esp/esp-idf/.git/modules/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib/HEAD /home/lee/esp/esp-idf/CMakeLists.txt /home/lee/esp/esp-idf/components/asio/asio/.git /home/lee/esp/esp-idf/components/bootloader/CMakeLists.txt /home/lee/esp/esp-idf/components/bootloader/project_include.cmake /home/lee/esp/esp-idf/components/bootloader/subproject/CMakeLists.txt /home/lee/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/CMakeLists.txt /home/lee/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc/.git /home/lee/esp/esp-idf/components/bootloader/subproject/main/CMakeLists.txt /home/lee/esp/esp-idf/components/bootloader_support/CMakeLists.txt /home/lee/esp/esp-idf/components/bt/controller/lib/.git /home/lee/esp/esp-idf/components/bt/host/nimble/nimble/.git /home/lee/esp/esp-idf/components/cbor/tinycbor/.git /home/lee/esp/esp-idf/components/coap/libcoap/.git /home/lee/esp/esp-idf/components/efuse/CMakeLists.txt /home/lee/esp/esp-idf/components/efuse/esp32/sources.cmake /home/lee/esp/esp-idf/components/esp32/CMakeLists.txt /home/lee/esp/esp-idf/components/esp32/project_include.cmake /home/lee/esp/esp-idf/components/esp_common/CMakeLists.txt /home/lee/esp/esp-idf/components/es [... cut ...]
./build/CMakeFiles/CMakeOutput.log:Run Build Command:"/home/lee/.espressif/tools/ninja/1.9.0/ninja" "cmTC_5100c"
./build/CMakeFiles/CMakeOutput.log:Run Build Command:"/home/lee/.espressif/tools/ninja/1.9.0/ninja" "cmTC_29c07"
./build/CMakeFiles/CMakeOutput.log:Run Build Command:"/home/lee/.espressif/tools/ninja/1.9.0/ninja" "cmTC_dffd1"
./build/rules.ninja:# It is included in the main 'build.ninja'.
./build/rules.ninja:  command = /home/lee/.espressif/tools/ninja/1.9.0/ninja -t clean
./build/rules.ninja:  command = /home/lee/.espressif/tools/ninja/1.9.0/ninja -t targets

```
Thanks 

Lee

kolipakakondal, 2021-02-16:
It's because CMakeCache.txt has old entries, it's suggested to delete the build folder and rebuild in case of project rename/duplicate

kolipakakondal, 2022-04-04:
Hello, All these problems were fixed with v2.4.0(latest v2.4.2). Please get the latest changes using the update site and report us back if you have any problems with this.

Thank you.

closed 2022-04-04 by kolipakakondal