espressif/idf-eclipse-plugin #72: Project has invalid configuration.

noweare1, 2020-04-12:
I am using esp-idf 4.0 and eclipse 2020-03 (4.15) and the latest plugin for eclipse.

I don't know if this is a bug or something that can be dealt with by
a work-around.

I have a esp32 project (project 2) that somehow got mis-configured in Eclipse.

I have another esp32 project (project 1) that works in eclipse as desired.

Project 1 has a launch
configuration as C/C++ project and the target is esp32.

Project 1 was initated as a new project inside
Eclipse.

Project 2 was an existing
project that was initiated from one of the template examples and
was built using the command line tools. It worked as desired.

I could not build Project 2 once
it was in eclipse. The error msg is "The project does not contain valid configurations
 
Project 2 launch configuration is "esp application" and the
target is "Local" (what ever that is, I don't know).

I have gone through the menus and any config files that I could find in the
project to see where I could change the configuration to be correct but
I am getting not getting much success.

I have tried deleting it from the workspace then importing it again but
I get the error "Project can not be created" Reason:Errors occurred while
refreshing resources with the local file system.

Also tried copying a working .cproject   to the non-working project 

Thanks for taking a look at this.

kolipakakondal, 2020-04-13:
Hello, Happy that it's working for you now.

> target is "Local" (what ever that is, I don't know).

esp-idf project doesn't work with the local target. You need to create an esp launch target. Please check this https://github.com/espressif/idf-eclipse-plugin#ConfigureLaunchTarget

> I have tried deleting it from the workspace then importing it again but
> I get the error "Project can not be created" Reason:Errors occurred while
> refreshing resources with the local file system.

Probably because some of the resources are deleted also from the file system.

To have a smooth workflow - please follow the guide until you feel comfortable.

kolipakakondal, 2020-04-13:
Closing the ticket as the issue got fixed, Please create a new ticket if you find any issue while working.

closed 2020-04-13 by kolipakakondal