espressif/idf-eclipse-plugin #73: debug ESP32 using JTAG

SERIDJ, 2020-04-15:
hi,

i have question about eclipse (windows 10 ) :
i can debug ESP32 using ESP32-Ethernet-Kit_A_V1.1 , board has JTAG interface already build in ?

[image]

There may be some small differences if i use windows 10. ?

i think Configuring Environment Variables, Configuring Toolchains and Configuring CMake Toolchain are detected well when i installed the tools using : Help > Espressif IDF Tools Manager > Install Tools option from the Eclipse.

[image]

[image]

[image]

my error : 

Error with command: gdb --version
Cannot run program "gdb": Launching failed
No esp launch target found. Please create/select the correct 'Launch Target'

[image]

kolipakakondal, 2020-04-15:
Hello @SERIDJ 

Can you check this thread https://github.com/espressif/idf-eclipse-plugin/issues/35#issuecomment-555878466

It seems to be an issue with the  xtensa-esp32-elf esp32-2019r1-8.2.0 version. Above thread will give you more details.

kolipakakondal, 2020-04-15:
To check the tools version details:

> Navigate to Help > Espressif IDF Tools Manager > List Installed Tools

SERIDJ, 2020-04-15:
thanks @kolipakakondal  i will check that and try again 

about version details i have ; 

Executing C:\Users\mohammed.seridj\AppData\Local\Programs\Python\Python38-32\python.exe C:\Users\mohammed.seridj\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

kolipakakondal, 2020-06-03:
Closing the issue as there is no update for quite some time.

We have released v1.1.0, please get the latest https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal