espressif/idf-eclipse-plugin #74: CMake is not picking up latest changes.

TJ-C, 2020-04-18:
**Describe the bug**
Changing the sdKonfig settings is not reflected in CMake.

**To Reproduce**
Steps to reproduce the behavior:
1. Double click on sdkonfig file
2. Make changes
3. Save
4. See error

**Expected behavior**
CMake step 1/5 creates the correct partition size.
CMake  step 2/3 uses correct parameters to flash device.

See screenshots and compare with console output below.

**Screenshots**
[image]
[image]

[1/5] cd /home/t/eclipse-workspace/gps/build/esp-idf/partition_table && /home/t/.espressif/tools/cmake/3.16.4/bin/cmake -E echo "Partition table binary generated. Contents:" && /home/t/.espressif/tools/cmake/3.16.4/bin/cmake -E echo "*******************************************************************************" && python /home/t/esp32/esp-idf/components/partition_table/gen_esp32part.py -q --offset 8000 --flash-size 4MB /home/t/eclipse-workspace/gps/build/partition_table/partition-table.bin && /home/t/.espressif/tools/cmake/3.16.4/bin/cmake -E echo "*******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x2f40,24K,
phy_init,data,phy,0x8f40,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] cd /home/t/esp32/esp-idf/components/esptool_py && /home/t/.espressif/tools/cmake/3.16.4/bin/cmake -D IDF_PATH="/home/t/esp32/esp-idf" -D ESPTOOLPY="python /home/t/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/t/eclipse-workspace/gps/build" -P /home/t/esp32/esp-idf/components/esptool_py/run_esptool.cmake
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool.py --chip esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 nmea_parser.bin
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1f40 for file: partition_table/partition-table.bin
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash 
cd /home/t/esp32/esp-idf/components/esptool_py && /home/t/.espressif/tools/cmake/3.16.4/bin/cmake -D IDF_PATH="/home/t/esp32/esp-idf" -D ESPTOOLPY="python /home/t/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/t/eclipse-workspace/gps/build" -P /home/t/esp32/esp-idf/components/esptool_py/run_esptool.cmake
ninja: build stopped: subcommand failed.
Executing action: flash
Running ninja in directory /home/t/eclipse-workspace/gps/build
Executing "ninja flash"...
ninja failed with exit code 1

**Environment Information:** 
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/4497520/eclipse_config.txt)

**ESP-IDF Tools Information:**
Executing python /home/t/esp32/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200406 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

TJ-C, 2020-04-18:
When I do get it to compile by cleaning the project. it constantly resets with this message:
`
rst:0x3 (SW_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7432
load:0x40078000,len:14836
load:0x40080400,len:4740
entry 0x400806a4
I (56) boot: ESP-IDF v4.2-dev-1126-gd85d3d969 2nd stage bootloader
I (56) boot: compile time 21:26:15
I (57) boot: chip revision: 1
I (61) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (77) qio_mode: Enabling QIO for flash chip WinBond
I (77) boot.esp32: SPI Speed      : 80MHz
I (79) boot.esp32: SPI Mode       : QIO
I (83) boot.esp32: SPI Flash Size : 4MB
I (88) boot: Enabling RNG early entropy source...
E (93) flash_parts: partition 0 invalid magic number 0x2df
E (99) boot: Failed to verify partition table
E (104) boot: load partition table error!
ets Jun  8 2016 00:22:57
`

kolipakakondal, 2020-05-08:
Thanks for reporting. It's an issue in the ESP-IDF. The team already identified and fixed the issue. We will update you once the fix is available in the ESP-IDF public release.

kolipakakondal, 2021-02-16:
Please verify with the latest esp-idf stable version, report us back if you see the same issue.

closed 2021-02-16 by kolipakakondal