espressif/idf-eclipse-plugin #75: Invalid ScannerInfoCache file causing Eclipse editor indexing issues

nvannote, 2020-04-19:
**Describe the bug**

I had a project that suddenly began marking standard fixed width integer types (those in stdint.h) as undeclared types in the Eclipse editor.  The project had previously been fine with no issues.

I eventually tracked it down to the following workspace project file.

“.metadata/.plugins/org.eclipse.cdt.core/infoCache/PROJECT_NAME/default.json”

* Where PROJECT_NAME is a placeholder of the project that was giving me problems.

At a high level; it appears this file is maintained in IDFBuildConfiguration.java.  Somehow this file becomes corrupt during a clean/build cycle.

To fix the issue one needs to exit Eclipse;  delete the json files in the infoCache directory; Restart Eclipse and force a compile with the build button (Cmd-B doesn’t seem to do it).

**To Reproduce**
Steps to reproduce the behavior:
1. Right click on a project and select "Clean".
2. Re/Build the project.
3. Observe some types in the editor being marked as undeclared. In my case it was types defined in stdint.h

**Expected behavior**
No editor indexing issues caused by a clean/build cycle.  Also expect a Cmd-B to be equivalent to pressing the build button on the toolbar.

**Screenshots**
Available on request.

**ESP-IDF Tools Information:**
Executing /usr/bin/python /Users/xxxxxxx/esp/esp-idf-v4.0/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

kolipakakondal, 2020-04-30:
Hi @nvannote Thanks for the detailed report. Will analyze to understand the root cause of the problem. Thanks

kolipakakondal, 2020-06-03:
These issues have been addressed in the v1.1.0 release, please update to the latest.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal