espressif/idf-eclipse-plugin #76: Eclipse is hanging at File->New->Expressif IDF project

shahharshit, 2020-04-24:
**Describe the bug**
Eclipse is hanging at File->New->Expressif IDF project. I have recently installed eclipse 2020-03. After that, I have followed steps for the "Installing IDF Plugins using update site url
" and "Installing ESP-IDF Tools" from [](https://github.com/espressif/idf-eclipse-plugin)

**To Reproduce**
Steps to reproduce the behavior:
1. Open eclipse. Go to Help -> Install New Software
2. Click Add… and Enter Location of the repository https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
3. Enter Name as Espressif IDF Plugins for Eclipse
4. Click Ok and Select Espressif IDF from the list and proceed with the installation
5. Navigate to "Help > Espressif IDF Tools Manager > Install Tools" 
2. Provide the ESP-IDF directory path "E:\Harshit\ESP-Projects\ESP32"
3. Provided path for "Git" bin "C:\Program Files\Git\bin\git.exe"
4. Clicked on " Install Tool" everything is installed.
5. Navigate to "Window > Perspective > Reset Perspective"
6. Navigate to "File > New > Espressif IDF Project"
7. Nothing is happening from here, eclipse hangs at here.

**Expected behavior**
From the steps mentioned in the [](https://github.com/espressif/idf-eclipse-plugin#create-a-new-project) it should ask for the project name.

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[eclipse-configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/4529173/eclipse-configuration.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
> Executing C:\Users\Harshit\AppData\Local\Programs\Python\Python37\python.exe E:\Harshit\ESP-Projects\ESP32\tools\idf_tools.py list
> ```
> * xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
>   - esp-2019r2-8.2.0 (recommended, installed)
> * esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
>   - 2.28.51.20170517 (recommended, installed)
> * cmake: CMake build system
>   - 3.13.4 (recommended, installed)
> * openocd-esp32: OpenOCD for ESP32
>   - v0.10.0-esp32-20190313 (recommended, installed)
> * mconf: menuconfig tool
>   - v4.6.0.0-idf-20190628 (recommended, installed)
> * ninja: Ninja build system
>   - 1.9.0 (recommended, installed)
> * idf-exe: IDF wrapper tool for Windows
>   - 1.0.1 (recommended, installed)
> * ccache: Ccache (compiler cache)
>   - 3.7 (recommended, installed)
> ```

kolipakakondal, 2020-04-28:
Hi @shahharshit 

Thanks for the log and detailed info. Could you also help me with the thread dump for the Eclipse?

You find the detailed info here https://wiki.eclipse.org/How_to_report_a_deadlock

On command line, use **jps -v** to find the PIDs of Java processes and jstack <pid> to show the stack trace of Java processes.

```
$ jps -v
32308 org.eclipse.equinox.launcher_1.5.0.v20180119-0753.jar -Dosgi.requiredJavaVersion=1.8 -Dosgi.instance.area.default=@user.home/eclipse-workspace -XX:+UseG1GC -XX:+UseStringDeduplication -Dosgi.requiredJavaVersion=1.8 -Dosgi.dataAreaRequiresExplicitInit=true -Xms256m -Xmx1024m -Dswt.dbus.init
8824 Jps -Dapplication.home=/usr/lib/jvm/java-8-oracle -Xms8m
$ jstack 32308 > /tmp/jstack.txt
```
Here 32308 and 8824 are PIDs of Java processes. 8824 is jps itself and is of no interest for us. 32308 is and Eclipse process judging from presence of org.eclipse.equinox.launcher in its command line. Jstack command saves stack trace of Eclipse process in a file /tmp/jstack.txt, attach the file to bug report.

shahharshit, 2020-04-30:
Hi @kolipakakondal,

Thank you for the quick response.

1) I have done trace dump using your mentioned method and attached here the jstack.txt
[jstack.txt](https://github.com/espressif/idf-eclipse-plugin/files/4556333/jstack.txt)

2) From the mentioned link I have followed steps given for getting the dump using "eclipse -consolelog" command for Windows.

> log4j:WARN No appenders could be found for logger (org.eclipse.jgit.internal.storage.file.FileSnapshot).                log4j:WARN Please initialize the log4j system properly.                                                                 !SESSION 2020-04-30 10:52:59.389 -----------------------------------------------                                        eclipse.buildId=4.15.0.I20200305-0155                                                                                   java.version=13.0.2                                                                                                     java.vendor=Oracle Corporation                                                                                          BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_IN                                                         Framework arguments:  -product org.eclipse.epp.package.cpp.product                                                      Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product -consolelog                                                                                                                              !ENTRY com.espressif.idf.core 1 0 2020-04-30 10:53:33.392                                                               !MESSAGE IDF path E:\Harshit\ESP-Projects\ESP32                                                                                                                                                                                                 !ENTRY com.espressif.idf.core 1 0 2020-04-30 10:53:33.789                                                               !MESSAGE IDF path E:\Harshit\ESP-Projects\ESP32

kolipakakondal, 2020-04-30:
Thanks for the thread dump @shahharshit

kolipakakondal, 2020-04-30:
@shahharshit It seems to be your **esp-idf** examples don't have the **hello_world** project. Could you check that?

For example:
`/Users/userName/esp/esp-idf/examples/get-started/hello_world`

However, I'll mark this as a bug to avoid blocking the eclipse.

shahharshit, 2020-04-30:
That's correct. I think the by default project is blick that they are giving. Should I just copy the new project hello_world and proceed further?

kolipakakondal, 2020-04-30:
https://github.com/espressif/esp-idf/tree/master/examples/get-started.

**hello-world** is part of the examples which you would get with esp-idf. Could you reset the esp-idf repo and see if you could get it?

shahharshit, 2020-04-30:
Yes, after "git reset --hard HEAD" getting the hello-world directory. Don't know somehow it got deleted. 
Thank you very much @kolipakakondal . I will try with other mentioned steps to compile and flash.

kolipakakondal, 2020-06-03:
The issue has been fixed with v1.1.0 release. Please update to the latest version.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal