espressif/idf-eclipse-plugin #77: Cannot build project in Eclipse

michaliskon, 2020-05-03:
I cannot build a copy of the project https_server

Console output:
	  

	Building in: C:\Users\User1\Dropbox\coding\Eclipse IDE C C++\HTTPS_server\build
	cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\User1\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 C:\Users\User1\Dropbox\coding\Eclipse IDE C C++\HTTPS_server
	-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
	-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
	-- Project version: 1
	-- Building ESP-IDF components for target esp32
	-- Checking Python dependencies...
	Python requirements from C:\Users\User1\esp-idf\requirements.txt are satisfied.

	CMake Error at C:/Users/User1/esp-idf/tools/cmake/component.cmake:222 (message):
	  CMake Error at
	  C:/Users/User1/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60
	  (include):

		include called with wrong number of arguments.  include() only takes one
		file.

	  Call Stack (most recent call first):

		C:/Users/User1/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

	Call Stack (most recent call first):
	  C:/Users/User1/esp-idf/tools/cmake/build.cmake:404 (__component_get_requirements)
	  C:/Users/User1/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
	  CMakeLists.txt:10 (project)

	-- Configuring incomplete, errors occurred!
	See also "C:/Users/User1/Dropbox/coding/Eclipse IDE C C++/HTTPS_server/build/CMakeFiles/CMakeOutput.log".
	cmake --build . -- -v
	ninja: error: loading 'build.ninja': The system cannot find the file specified.

	Build complete (0 errors, 0 warnings): C:\Users\User1\Dropbox\coding\Eclipse IDE C C++\HTTPS_server\build

Also I noticed that all the libraries included at the top of main are "unresolved inclusion".

Apparently I have not set something correctly, but I cannot figure out what it is.

kolipakakondal, 2020-05-04:
> include called with wrong number of arguments.  include() only takes one
> 	file.

Hello, It seems to be more of a program error. Check if you can identify and fix this. Let us know if you need more help with this.

"unresolved inclusion" errors will disappear after the build completion.

michaliskon, 2020-05-04:
Hi, I haven't used eclipse before and I do not know how to proceed.

I would like to request some more help with this.

kolipakakondal, 2020-05-04:
@projectgus @igrr Could you please help him with this error. Thanks

kolipakakondal, 2020-05-04:
> Hi, I haven't used eclipse before and I do not know how to proceed.
> 
> I would like to request some more help with this.

Could you please share a sample project to reproduce this error.

michaliskon, 2020-05-04:
I have also zipped it and uploaded it.
It is the https_server under protocols examples from esp-idf.

[HTTPS_server.zip](https://github.com/espressif/idf-eclipse-plugin/files/4573924/HTTPS_server.zip)

kolipakakondal, 2020-05-04:
> I have zipped it and uploaded it.
> It is the https_server under protocols examples from esp-idf.
> 
> [HTTPS_server.zip](https://github.com/espressif/idf-eclipse-plugin/files/4573924/HTTPS_server.zip)

I'm able to build the same project you've provided without any compilation errors. The problem could be with the CMake version or something else. Will update you.

michaliskon, 2020-05-04:
> I'm able to build the same project you've provided without any compilation errors. The problem could be with the CMake version or something else. Will update you.

I am looking forward to.

projectgus, 2020-05-04:
Hi @michaliskon ,

At the moment ESP-IDF doesn't support spaces in the path to ESP-IDF or the project. 

> C:\Users\User1\Dropbox\coding\Eclipse IDE C C++\HTTPS_server\build

If you remove the spaces in "Eclipse IDE C C++" and reopen & rebuild the project, then it should work.

Sorry for the inconvenience. We plan to revisit this in the near future.

michaliskon, 2020-05-05:
Thank you. It works fine now :)

closed 2020-05-05 by michaliskon

kolipakakondal, 2020-05-05:
Thanks @projectgus. Good to know.