espressif/idf-eclipse-plugin #78: ESP-IDF plugin webpage not reachable

hoi44, 2020-05-08:
**Describe the bug**
The Eclipse ESP-IDF plugin webpage is not reachable. https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ is returning 403.

**To Reproduce**
One of the following methods reproduce the error:
* Eclipse -> Help -> Install New Software … -> Add … -> https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
* Eclipse -> Help -> Eclipse Marketplace … -> ESP-IDF Eclipse plugin v1.0.1
* call GET https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

**Expected behavior**
That the plugin would install itself correctly.

**Screenshots**
[image]
<details><summary>Error code (see below)</summary>
<p>
```
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
Unable to read repository at https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/content.xml.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
```
</p>
</details>

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
<details><summary>Details (see below)</summary>
<p>
```
-ws
win32
-arch
x86_64
os.arch=amd64
os.name=Windows 10
os.version=10.0
osgi.arch=x86_64
-showsplash
[...] eclipse\\plugins\org.eclipse.epp.package.common_4.15.0.20200313-1200\splash.bmp
-launcher
[...] eclipse\eclipse.exe
--launcher.library
[...] eclipse\\plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.1.1100.v20190907-0426\eclipse_1902.dll
-startup
[...] eclipse\\plugins/org.eclipse.equinox.launcher_1.5.700.v20200207-2156.jar
-vm
C:\Program Files\Java\jre1.8.0_191\bin\server\jvm.dll
```
</p>
</details>

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
unfortunately not present

kolipakakondal, 2020-05-08:
This would have caused by the Java version or Java certificates. Please make sure you've installed Java 8 Update 212 and later to fix this error.
Check below links:

https://esp32.com/viewtopic.php?f=13&t=12327&start=10#p50137
https://stackoverflow.com/questions/6908948/java-sun-security-provider-certpath-suncertpathbuilderexception-unable-to-find

hoi44, 2020-05-13:
Thx for your response.
I have Java 8 Update 191 installed, which I can't update (no admin rights).
Even with a newly installed VM (ESXi, Ubuntu 20.04 LTS, Eclipse 2020‑03, newest JDK version) I get the same error.
Meanwhile I tried using my Android Phone, which also shows the same error (403).
Since I have no admin rights, I also can't install certificates.
NAT should have nothing to do with Java, my guess is that the server has changed some permission rights.

Are you sure that https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ is accessable without further Java/Certificates configuration?

kolipakakondal, 2020-05-14:
Hi @hoi44 

If you can install Java 8 Update 212 and above(even the latest version Java) to fix the issue. Plugins won't work anything below that version. I'll update the document accordingly.

Since we have not added any webpage to the update site you're seeing **403 Forbidden** page  but that's nothing to do with installing the plugins.

nvannote, 2020-06-03:
> Hi @hoi44
>
> If you can install Java 8 Update 212 and above(even the latest version Java) to fix the issue. Plugins won't work anything below that version. I'll update the document accordingly.
>
> Since [...]

I am on macOS 10.14.6 and having the exact same issue.  With the exception that I do have Java 8 Update 251 installed (latest available).

Trying to update the plugin from v1.0.1 to v1.1.0.

Faced with the same errors previously;  I was able to get the v1.0.1 plugin installed with a local install (not sure where I obtained that image).  But I can't seem to locate a plugin image for v1.1.0 to perform a similar install.

I will note that the Espressif ESP-IDF plugin for Eclipse is the only update site that has been causing problems;  all my other Eclipse update sites work perfectly fine.

Best Regards

kolipakakondal, 2021-02-16:
As the original issue reported here is because of the Java version in the user machine, hence closing the issue

closed 2021-02-16 by kolipakakondal

kolipakakondal, 2021-10-06:
Hi @DomnTMi 

I would request you to change your eclipse version to `Eclipse 2021-06` as we have not added support for `Eclipse 2021-09`(the one which you've used). Open bug is [here](https://github.com/espressif/idf-eclipse-plugin/issues/363)

If you are having issues with the update site, could you use the[ zip file](https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-v2.2.0.zip) to download and update it 

Please refer these
https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0

Thanks