espressif/idf-eclipse-plugin #79: Adding custom and other libraries in Eclipse ESP-IDF

michaliskon, 2020-05-08:
I have only recently started using Eclipse and ESP-IDF.
I have made and uploaded several programs using the ESP-IDF built-in libraries without problem.

I now want to use my custom libraries and I do not really now how to instruct eclipse to include them.
I am also trying unsuccessfully to use an external library for the LCD of the WROVER-KIT. [https://github.com/espressif/esp-iot-solution/tree/master/components/spi_devices/lcd](url) or [https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_master/lcd](url)

I've been looking on instructions how to do that with eclipse, but I am unsuccessful. Mostly because some settings do not seem to appear. (i.e. there is no "C/C++ build" in Project Properties, neither "Paths and Symbols" in "Project Properties > C/C++ General")

I have two questions:
a) How do I configure eclipse (or the compiler) to grab my custom library or libraries from a certain path in the disk, so I can #include "Custom_library.c" on my project and use the functions I have in there?
b) Is it the same way to use one of the LCD libraries if I save it somewhere on my disk? Probably close to my custom libraries?

kolipakakondal, 2020-05-09:
Here is the example: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib

Basically, this part: https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/import_lib/main/CMakeLists.txt#L19-L23

You can also find **build_system** example while creating a new esp-idf project using eclipse.

michaliskon, 2020-05-09:
> 
> 
> Here is the example: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib
> 
> Basically, this part: https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/import_lib/main/CMakeLists.txt#L19-L23
> 
> You can also find **build_system** example while creating a new esp-idf project using eclipse.
> 
> Old Eclipse CDT use to have settings in the preferences to achieve this but that got removed in the recent versions of Eclipse. I would suggest to use the above CMake based approach to include the external lib's to the project and rebuild it. It will automatically resolve the headers and inclusion errors.

From what I understand, I need to change the contents of the main\CMakeLists.txt

I can sort of see the logic behind this, but I do not really understand how to apply this. For example my CMakeLists.txt is different than the one in build_system example. Do I simply append the add_subdirectory(...) and target_link_libraries(...)? Does something similar to the _${COMPONENT_LIB}_ in the example required to be added in Eclipse CDT paths?

Could you help me with a specific project telling what steps are needed to add the _tftspi.h_, _tft.h_ and _spiffs_vfs.h_ that are under _components_ and _My_custom_lib.c_ which is in _C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\My_custom_lib_?
I would help me a lot understand the logic.

[https://i.ibb.co/Btqnv6q/Eclpise.png](url)

Also, do I modify manually the CMakeLists.txt or is there something in the IDE that is autogenerating it?

kolipakakondal, 2020-05-12:
@projectgus Could you please help with this.

michaliskon, 2020-05-17:
Hi again, I still haven't found a way to solve this, except from adding the whole path in the main.cm for example: ` C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\My_WiFi\My_WiFi.c` and ` C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\My_UART\My_UART.c`
I am not sure if that will suffice once I want to include library for some hardware, usually they include multiple files.

Is there a way to add `C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\` to the compiler paths, so the compiler can search for the files `My_UART.c` and `My_WiFi.c` in there?

projectgus, 2020-05-18:
Hi @michaliskon ,

Sorry for the delay in getting back to you.

Additional libraries in ESP-IDF are usually included as separate "components", for example [the lcd component which you linked to inside the iot-solution repo](https://github.com/espressif/esp-iot-solution/tree/master/components/spi_devices/lcd).

Adding this to your project can be done by making a directory name "components" inside your ESP-IDF project directory (ie same directory as your project's top-level CMakeLists.txt file). Then copy the "lcd" directory from the IoT solution into that components directory (ie `<project>/components/lcd/`).

You can see a complete example of a project layout, here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#example-project

After you add these components into your project, the C source files in your project's `main` component should be able to include the headers from the other components directly. Note that if you want to include one component's headers from another component (apart from the "main" component), [a 'requirement' relationship needs to be set up between the two](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements).

Unfortunately there isn't an Eclipse UI to manage this at the moment, it's all done by moving files into place and editing the CMakeLists.txt files directly.

michaliskon, 2020-05-18:
Hi @projectgus, thank you for the replay.

- Is there a way to link the lcd library as a component if located another place? so it can be linked to several projects without copying it?
- Is there a way to link a path that includes my own library files as descrived below?

> 
> 
> Hi again, I still haven't found a way to solve this, except from adding the whole path in the main.c for example:
> `#include "C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\My_WiFi\My_WiFi.c"`
> and
> `#include "C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\My_UART\My_UART.c"`
> 
> I am not sure if that will suffice once I want to include library for some hardware, usually they include multiple files.
> 
> Is there a way to add `C:\Users\User1\Dropbox\coding\ESP-IDF\libraries\` to the compiler paths, so I can:
> `"#include My_UART.c"`
> `"#include My_WiFi.c"`
> and the compiler can search for the files?

michaliskon, 2020-05-23:
Hi,

I am still looking for an answer to this problem. 

It's pretty basic in other development systems. 

I'm quite surprised if it is not possible in eclipse esp-idf.

projectgus, 2020-05-25:
Hi @michaliskon ,

Sorry somehow I missed the notification for your reply.

>  Is there a way to link the lcd library as a component if located at another place? so it can be linked to several projects without copying it?

Yes, set the `EXTRA_COMPONENT_DIRS` variable in your project's CMakeLists.txt file. Set it to the directory containing the lcd library component, or the parent of that directory (if you want the project multiple components under the parent directory).

The value of this variable can be a relative path (relative to the project directory) or an absolute path.

Something like this is done in some ESP-IDF examples to include common example connection boilerplate:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/CMakeLists.txt#L7

>    Is there a way to link a path that includes my own library files as descrived below?

Yes. You can either convert each library to its own component, or create a single component and list all the relevant libraries' source files and header directories in that component.

vitaliy2034, 2020-10-18:
To add custom folder path you need write additional line to **"<project_path>\main\CMakeLists.txt"** 
`set(COMPONENT_ADD_INCLUDEDIRS "Path_ to_incdir")` 
[image]

closed 2021-02-16 by kolipakakondal