espressif/idf-eclipse-plugin #81: ESP-IDF 仅适用于 Python 2.7， 而这里要求  Python 3.5 and above

hiliuyj, 2020-05-12:
**Describe the bug**
ESP-IDF 编程指南 (最新版) ：  设置 Python 解释器
目前，ESP-IDF 仅适用于 Python 2.7，如果系统中默认的 python 解释器是 Python 3.x，可能会出现问题。
而 Espressif IDF Eclipse Plugins， README.md ， Installing Prerequisites ， Python 3.5 and above， 
到底 要用 哪个版本 Python  ？

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

igrr, 2020-05-12:
@hiliuyj ESP-IDF 4.x versions support Python 3.5 and above. You can choose the Python 3 interpreter in the plugin. If you have met some issue, please provide more detailed information, as requested in the issue template.

hiliuyj, 2020-05-13:
不知道是否 prthon 版本原因， 
出错 :  使用 ESP-IDF V4.2 , 在 Eclipse (版本 2020-03) 中 使用  Python 3.6.9 ,  以 Hello_world 例子 为模板 生成 一个 项目， 编译 后， 编辑 配置文件， 报错如下：Traceback (most recent call last):
  File "/home/hiliu/esp/esp-idf/tools/idf.py", line 775, in <module>
    main()
  File "/home/hiliu/esp/esp-idf/tools/idf.py", line 692, in main
    cli = init_cli(verbose_output=checks_output)
  File "/home/hiliu/esp/esp-idf/tools/idf.py", line 650, in init_cli
    extensions[name] = import_module(name)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 888, in get_code
  File "<frozen importlib._bootstrap_external>", line 455, in _validate_bytecode_header
ImportError: bad magic number in 'dfu_ext': b'\x03\xf3\r\n'

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our latest builds

closed 2021-02-16 by kolipakakondal