espressif/idf-eclipse-plugin #82: 'unrecognized arguments: --no-site-packages' during esp-idf tools installation

ahinsen, 2020-05-15:
Hi, 
I freshly downloaded and installed esp-idf 4.0 on Win10 according to the [documentation](https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html#get-started-get-esp-idf) I could successfully flash and monitor the hello world example from command line.
Then I wanted to do the same in eclipse, so I proceeded by the readme.md of this project.
At 'Installing ESP-IDF Tools' the console shows this error:
```
Executing C:\Program Files (x86)\Python38-32\python.exe C:\DEV\esp-idf40\dl\tools\idf_tools.py install-python-env
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                  [--clear] [--system-site-packages] [--copies] [--no-download | --download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                  [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Creating a new Python environment in C:\DEV\esp-idf40\python_env\idf4.0_py3.8_env
Traceback (most recent call last):
  File "C:\DEV\esp-idf40\dl\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "C:\DEV\esp-idf40\dl\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "C:\DEV\esp-idf40\dl\tools\idf_tools.py", line 1182, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', '--no-site-packages', idf_python_env_path],
  File "C:\Program Files (x86)\Python38-32\lib\subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Program Files (x86)\\Python38-32\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\DEV\\esp-idf40\\python_env\\idf4.0_py3.8_env']' returned non-zero exit status 2.
```
**To Reproduce**
1. Delete the 'python-env' folder from your 'esp-idf' folder, and make sure you have Python38-32 installed
2. Install ESP-IDF Tools as described in the readme.md
3. See the console

**Expected behavior**
successfull installation, and an item showing up in eclipse 'Help/Espressive IDF tools manager/List installed tools'

**Screenshots**
[image]

**Environment Information:** 
[eclipseConfig.txt](https://github.com/espressif/idf-eclipse-plugin/files/4636613/eclipseConfig.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools ---> this list is empty :(

Any help is welcome - Thanks

igrr, 2020-05-15:
Hi @ahinsen, this is an issue in v4.0 release, which has a fix on release/v4.0 branch (see https://github.com/espressif/esp-idf/issues/5204#issuecomment-620537960). Please try one of the following:

- run the installer again, this time choosing "release/v4.0" version of IDF to download
- update the existing copy of IDF using `git checkout release/v4.0`, `git pull origin release/v4.0`.

ahinsen, 2020-05-16:
Thanks @igrr, it works without error, but 'Help/Espressif IDF tools Manager/List installed tools' still does nothing. Should I be worried about that?

kolipakakondal, 2020-05-18:
Hi @ahinsen, Could you check the error log to see the reported error https://github.com/espressif/idf-eclipse-plugin#error-log

ahinsen, 2020-05-18:
Hi @kolipakakondal, Indeed: It seems the installer didn't set the python path. I attach a log of just two actions: install tools, list tools. I also attach a screenshot where I supplied the python path. Could you please have a look at it? Thanks for your help!
[inst-tools.log](https://github.com/espressif/idf-eclipse-plugin/files/4642389/inst-tools.log)
[image]

vsecades, 2020-09-24:
> Hi @ahinsen, this is an issue in v4.0 release, which has a fix on release/v4.0 branch (see [espressif/esp-idf#5204 (comment)](https://github.com/espressif/esp-idf/issues/5204#issuecomment-620537960)). Please try one of the following:
> 
> * run the installer again, this time choosing "release/v4.0" version of IDF to download
> * update the existing copy of IDF using `git checkout release/v4.0`, `git pull origin release/v4.0`.

Thanks, this worked :-). Update the docs when possible.

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release.

closed 2021-02-16 by kolipakakondal