espressif/idf-eclipse-plugin #83: Build fail - undefined reference to

kifantidis, 2020-05-21:
Hello there. 
I've recently installed eclipse and esp-idf plugin. I've compiled/flashed/debugged couple of examples successfully. But when I'm trying a project that includes wifi the ninja tool can't build the project. When I delete this command: esp_wifi_init(&cfg); everything works again. I think this problem is related to component includes and paths because before ninja stops building the program I get some "undefined reference to" messages (Ive posted them bellow. Build-LogFile). 

Is there anyone that has or had the same problem ? 

> **In order to reproduce this:**
> Open eclipse. Go in C/C++ prespective 
> File-->Import-->Espressif-->Espressif IDF Project-->Next. 
> Set project name and location and press "Next" again.
> Choose wifi smartconfig example.
> Build the example

> 
> **Environment**
>         Development Kit: ESP-WROVER-KIT
>         Kit version: v3
>         Module or chip used: ESP32-WROVER
>         IDF version: release/v4.0
>         Build System: Cmake
>         Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
>         Operating System: Windows
>         Using an IDE?: Eclipse + esp-idf plugin
>         Power Supply: USB
> 

```

**Build-LogFile:**
[6/7] cmd.exe /C "cd . && C:\Users\virtual_pc\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\smart_config.elf.rsp  -o smart_config.elf  && cd ."
FAILED: smart_config.elf 
cmd.exe /C "cd . && C:\Users\virtual_pc\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\smart_config.elf.rsp  -o smart_config.elf  && cd ."
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa_main.c.obj):(.literal.wpa_config_profile+0xc): undefined reference to `esp_wifi_sta_prof_is_wpa3_internal'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa_main.c.obj): in function `wpa_config_profile':
C:/esp/esp-idf/components/wpa_supplicant/src/esp_supplicant/esp_wpa_main.c:77: undefined reference to `esp_wifi_sta_prof_is_wpa3_internal'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj):(.literal.ieee80211w_set_keys+0x0): undefined reference to `esp_wifi_set_igtk_internal'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj):(.literal.wpa_set_bss+0x4): undefined reference to `esp_wifi_sta_pmf_enabled'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj):(.literal.wpa_set_bss+0x8): undefined reference to `esp_wifi_sta_get_mgmt_group_cipher'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `ieee80211w_set_keys':
C:/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:1011: undefined reference to `esp_wifi_set_igtk_internal'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `wpa_set_bss':
C:/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:2122: undefined reference to `esp_wifi_sta_pmf_enabled'
c:/users/virtual_pc/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:2130: undefined reference to `esp_wifi_sta_get_mgmt_group_cipher'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (11 errors, 0 warnings): C:\esp\01_projects\test_wifi\build

```

kolipakakondal, 2020-05-21:
@igrr @projectgus Any inputs for this?

kifantidis, 2020-05-21:
Hello again.
Updating esp-idf in version v4.0.1 fixed the problem. 

`git clone -b v4.0.1 --recursive https://github.com/espressif/esp-idf.git`

closed 2020-05-21 by kifantidis