espressif/idf-eclipse-plugin #84: Issue with intallation of virtualenv.

Ymoris, 2020-05-21:
Issue with intallation of virtualenv.

1) Hardware: AMD64
2) Operating system; Linux with Ubuntu 20.04
3) Eclipse
	Version: 2019-12 (4.14.0)
	Build id: 20191212-1212

4) ESP IDF package installed

5) Installed tolls:

	Executing python /home/user/esp-idf-v4.0/tools/idf_tools.py list
	* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  	- esp-2019r2-8.2.0 (recommended, installed)
	* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  	- 2.28.51.20170517 (recommended, installed)
	* cmake: CMake build system (optional)
  	- 3.13.4 (recommended, installed)
	* openocd-esp32: OpenOCD for ESP32
  	- v0.10.0-esp32-20190313 (recommended, installed)
	* ninja: Ninja build system (optional)
  	- 1.9.0 (recommended, installed)
  	
 6) Issue with install virtualenv 
 
 	Executing python /home/user/esp-idf-v4.0/tools/idf_tools.py install-python-env
 	
 	virtualenv: error: unrecognized arguments: --no-site-packages
 	
 	Creating a new Python environment in /home/user/.espressif/python_env/idf4.0_py3.8_env
 	
 	Traceback (most recent call last):
  	File "/home/user/esp-idf-v4.0/tools/idf_tools.py", line 1372, in <module>
    		main(sys.argv[1:])
  	File "/home/user/esp-idf-v4.0/tools/idf_tools.py", line 1368, in main
    		action_func(args)
  	File "/home/user/esp-idf-v4.0/tools/idf_tools.py", line 1183, in action_install_python_env
    		subprocess.check_call([sys.executable, '-m', 'virtualenv', '--no-site-packages', idf_python_env_path],
  	File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    		raise CalledProcessError(retcode, cmd)
	subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--no-site-packages', '/home/user/.espressif/python_env/idf4.0_py3.8_env']' returned non-zero exit status 2.

Executing python /home/user/esp-idf-v4.0/tools/idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts
PATH=/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/user/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/user/.espressif/tools/cmake/3.13.4/bin:/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/home/user/.espressif/tools/ninja/1.9.0/:/home/user/esp-idf-v4.0/tools:$PATH

7) Configuring Environment Variables and adding PATH

	Click on the Environment preference page under C/C++ Build.
	
	The menu is:
	Build Targets
	Console
	Logging
	Makefile Editor
		Settings
	Settings
	CMake
	
	The Environment submenu to access the PATH is not available.

8) The Arduino IDE is also used.

kolipakakondal, 2020-05-21:
Please check this https://github.com/espressif/idf-eclipse-plugin/issues/82#issuecomment-629443827

Ymoris, 2020-05-22:
Thanks for the answer.

I install the version 4.0.1.

The installation has been successful.

This the console content:
Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py install-python-env
Requirement already satisfied: setuptools in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 4)) (46.4.0)
Requirement already satisfied: click>=5.0 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (2.9.2)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: six>=1.4.1 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (2.20)
Installing Python packages from /home/user/esp-idf-v4.0.1/requirements.txt

Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts
IDF_PYTHON_ENV_PATH=/home/user/.espressif/python_env/idf4.0_py3.8_env
PATH=/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/user/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/user/.espressif/tools/cmake/3.13.4/bin:/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/home/user/.espressif/tools/ninja/1.9.0/:/home/user/.espressif/python_env/idf4.0_py3.8_env/bin:/home/user/esp-idf-v4.0.1/tools:$PATH

But a new IDF project could not be created.

The Eclipse workspe is home/user/eclipse-workspace.

The Finish button is inactive.

Is it necessary to uninstall and re-install the Eclipse ESP-IDF package?

closed 2020-05-26 by kolipakakondal