espressif/idf-eclipse-plugin #85: [Ubuntu 18.04] Project create/import hangs

jmg86, 2020-05-26:
**Describe the bug**
When I try to create a new "Espressif IDF project" I cannot "finish" the creation, the button doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. File > New > project
2. choose "Espressif IDF project", then "next"
3. project name : "test", use default location and "next"
4. select "create the project using one of the templates", choose "Hello world" and press "finish"

** Observed behavior **
Nothing happens, the windows doesn't close and no project is created.

**Expected behavior**
The window should close and a new project should be created.

**Screenshots**
-

** More details **
1. Error log shows an error in "org.eclipse.ui.workbench" plugin with "Unhandled event loop exception message".
In the exception stack trace I found "Caused by: java.lang.ClassNotFoundException: javax.xml.bind.JAXBException cannot be found by org.eclipse.tools.templates.freemarker_1.1.100.202001140305"
[eclipse_error.txt](https://github.com/espressif/idf-eclipse-plugin/files/4680887/eclipse_error.txt)

2. I suspected the OpenJDK I used and replaced it with official Java from Oracle, but still have the same behavior.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/4680861/eclipse_config.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
Executing python /home/jmg/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200420 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

kolipakakondal, 2020-05-26:
Hi @jmg86, Thanks for reporting the issue.

I couldn't reproduce the issue with **Java 11.0.7** on **macOS X**.

I'm thinking the problem could be with the **JAVA 11** on **Linux** https://bugs.eclipse.org/bugs/show_bug.cgi?id=536316

Before I make any changes, could you please get the latest version of Java 13/14 and see if it works?

Thanks

jmg86, 2020-05-26:
Hi @kolipakakondal 
Thanks for your answer.
I've just tried with JavaSE 14.0.1+7 but unfortunately the problem remains the same.
Please note that my problem seems to be the same than Ymoris report.
Regards

kolipakakondal, 2020-05-27:
Hi @jmg86, Could you help me to verify this fix with the below local build
com.espressif.idf.update-v1.0.1-66-gcb584069.zip

Please follow these instructions https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

jmg86, 2020-05-27:
Hi @kolipakakondal 
I'm sorry but I can't find the zip file you mention.
Regards

kolipakakondal, 2020-05-27:
Sorry, fixed now. Here is the one

[com.espressif.idf.update-v1.0.1-66-gcb584069.zip](https://github.com/espressif/idf-eclipse-plugin/files/4682717/com.espressif.idf.update-v1.0.1-66-gcb584069.zip)

jmg86, 2020-05-27:
Hi @kolipakakondal 
The plugin you provided fixes the problem, at least for a project creation following the scheme I described.
Many thanks.
Regards

kolipakakondal, 2020-06-03:
Fixed java 11 issue with v1.1.0 release. Please update to the latest.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal