espressif/idf-eclipse-plugin #86: A new IDF project could not be created.

Ymoris, 2020-05-26:
A new IDF project could not be created.

The Eclipse workspe is home/user/eclipse-workspace.

The Finish button is inactive.

Is it necessary to uninstall and re-install the Eclipse ESP-IDF package?

The installation has been successful.

This the console content:
Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py install-python-env
Requirement already satisfied: setuptools in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 4)) (46.4.0)
Requirement already satisfied: click>=5.0 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (2.9.2)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from -r /home/user/esp-idf-v4.0.1/requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: six>=1.4.1 in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in /home/user/.espressif/python_env/idf4.0_py3.8_env/lib/python3.8/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/user/esp-idf-v4.0.1/requirements.txt (line 11)) (2.20)
Installing Python packages from /home/user/esp-idf-v4.0.1/requirements.txt

Executing python /home/user/esp-idf-v4.0.1/tools/idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts
IDF_PYTHON_ENV_PATH=/home/user/.espressif/python_env/idf4.0_py3.8_env
PATH=/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/user/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/user/.espressif/tools/cmake/3.13.4/bin:/home/user/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/home/user/.espressif/tools/ninja/1.9.0/:/home/user/.espressif/python_env/idf4.0_py3.8_env/bin:/home/user/esp-idf-v4.0.1/tools:$PATH

kolipakakondal, 2020-05-26:
Hi @Ymoris 

Could you send us the error log? Please follow the instructions from here https://github.com/espressif/idf-eclipse-plugin#error-log

Ymoris, 2020-05-26:
[Untitled.log](https://github.com/espressif/idf-eclipse-plugin/files/4682346/Untitled.log)

kolipakakondal, 2020-05-26:
Hi @Ymoris, Could you help me to verify this fix with the below local build
[com.espressif.idf.update-v1.0.1-66-gcb584069.zip](https://github.com/espressif/idf-eclipse-plugin/files/4682717/com.espressif.idf.update-v1.0.1-66-gcb584069.zip)

Please follow these instructions https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

Ymoris, 2020-05-26:
Creation of a new project is OK
testblink project has been created from the example blink.
Terminal could be selected /dev/ttyUSB1 and message from the connected ESP32 could be received.

But could not found any ESP target. 

Click on the third dropdown
Select New Launch Target
Select ESP Target

kolipakakondal, 2020-05-27:
Check if this helps https://medium.com/@kondal.kolipaka/not-found-your-serial-port-for-your-esp-wrover-kit-board-7b9696d123c1

kolipakakondal, 2020-05-27:
> Creation of a new project is OK

So you're able to create a new project using the local build which I've provided?

Ymoris, 2020-05-27:
Yes, I am able to create a new project using the local build. 

The serial port is also available /dev/ttyUSB1.
If a terminal is opened, then I could receive a message from a pre-loaded program.

But, 

Select New Launch Target
Select ESP Target

is not available.

kolipakakondal, 2020-05-27:
Hi @Ymoris, I'm sorry, the issue is not very clear for me. You meant to say serial port **/dev/ttyUSB1** is not available in the new ESP Target wizard, but the same serial port is available in the Serial Terminal window?

Ymoris, 2020-05-27:
Sorry too.

 In fact ESP target wizard is not installed or accessible.

Yes, the serial port is available in the Serial Terminal window. I can open the Terminal dialogue box and setup the serial port. The Terminal could receive text from the connected ESP32 at /dev/ttyUSB1.

I can not compile and download the project to the ESP32 target. No icon or dialogue box related to an ESP target Wizard.

kolipakakondal, 2020-05-28:
If I understand correctly, you're not able to see Launch Target drop-down itself? Maybe you can share a screenshot/short video about the problem?

You can try the following options in combination and see if it helps.

- Make sure you're in the **C/C++** perspective
- Reset perspective
- Try to create a new project and see if you're able to see three drop-downs on the toolbar
- Restart eclipse

Ymoris, 2020-05-28:
If I understand correctly, you're not able to see Launch Target drop-down itself? Yes

I proceed with the  options and I can not see the launch Target drop-down itself.

Note: I am a beginner with Eclipse.

[image]

kolipakakondal, 2020-05-29:
That's an interesting issue.

Can you make sure "**Enable the Launch Bar**" is checked in the preferences?
[image]

Ymoris, 2020-05-29:
This is the launch preferences:

[image]

Perspective is C/C++.

kolipakakondal, 2020-05-29:
How did you install Eclipse CDT? Looks like you've not installed the Eclipse CDT package as a whole. Probably, you would have updated the CDT on top of the standard Eclipse and forgot to select the LaunchBar feature.

As we mentioned here https://github.com/espressif/idf-eclipse-plugin#installing-prerequisites
> Eclipse 2018-12 CDT and above : Download and install Eclipse CDT package from here

Could you install this one https://www.eclipse.org/downloads/packages/release/2020-03/r/eclipse-ide-cc-developers-includes-incubating-components

kolipakakondal, 2020-05-29:
I could see CDT LaunchBar plugins in the log which you've provided. The issue might be something else. 

Anyway, please install the Eclipse CDT which I provided above, and see if it helps.

Ymoris, 2020-05-29:
After installing Eclipse CDT. This the screen shot.

The Launch  bar is there, but no Launch Configurations available.

[image]

Ymoris, 2020-05-29:
After installing ESP tool chain with the provided com.espressif.idf.update-v1.0.1-66-gcb584069.zip.

The configuration is available.

I can build, but can not Run.

It seems that the serial port creates a issue. 

Note; The connected ESP32 is running a Arduino sketch.

See screen capture.

[image]

Ymoris, 2020-05-29:
Finally, I reset the port to USB0 and get it Run.

The minor problem: The port is not retained in the terminal dialog box. Need to be rewite before run.

Thanks for your support.

This is the screen shot:
[image]

kolipakakondal, 2020-06-03:
Fixed java 11 issue with v1.1.0 release. Please update to the latest.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.0

closed 2020-06-03 by kolipakakondal