espressif/idf-eclipse-plugin #88: 'Run As'/Execute program with varargs

rherrmannr, 2020-05-27:
Hey ,
is there a way to execute the program using varargs?

I've changed the app_main function to:
```c
void app_main(int argc, char **argv) {
}
``` 
and hoped to use the varargs.

Adding arguments to the run config does not work..
[image]
as the python script seems to expect files:
```
/home/rherrmann/.espressif/python_env/idf4.0_py2.7_env/bin/python: can't open file '192.168.0.1': [Errno 2] No such file or directory
```

I couldn't find anything in the sdkconfig file.

kolipakakondal, 2020-05-29:
@igrr  @projectgus - Can we pass the arguments to the app_main(..)? Could you please help on this.

igrr, 2020-05-29:
Hi @rherrmannr, in ESP-IDF the app_main function is called without arguments, you can see the call here: https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/components/esp32/cpu_start.c#L580

When the program runs on the ESP32, it is loaded and executed by the bootloader, the process is explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#application-startup-flow). At this point, there is no communication between the ESP32 and the PC, other than that the ESP32 logs some of the events to the serial console.

Generally there can be two ways to pass arguments to the application (although not to `app_main` function, as its signature doesn't include any arguments):

1. pass the arguments from the PC
2. store the arguments in Flash and let the program read them

The first option can be implemented in a couple of ways:

1. JTAG semihosting, where the application executes a semihosting call to the PC, requesting the argument list. We do support semihosting ([example](https://github.com/espressif/esp-idf/tree/master/examples/storage/semihost_vfs)), but don't yet support the specific semihosting call used to fetch function arguments. I'll file that as a feature request in ESP-IDF.
2. Reading the arguments from UART: on startup, the application can read the argument list from UART, and then split the line into arguments. This can be done using a [console](https://github.com/espressif/esp-idf/tree/master/examples/system/console), or by reading the line manually and calling the [esp_console_split_argv](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/console.html#_CPPv422esp_console_split_argvPcPPc6size_t) function.

The second option (storing the arguments in Flash) can be implemented using the [NVS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html) component. You can first use the [NVS partition generator utility](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#nvs-partition-generator-utility) to generate a partition binary with the required key-value pairs inside, flash the partition binary into the device, and then read the values from your program at run time.

Unfortunately, all of these options are more involved than just adding launch arguments in Eclipse.

closed 2026-01-22 by AndriiFilippov