espressif/idf-eclipse-plugin #89: Unresolved Inclusion on base installation of ESP-IDF on Eclipse on Windows 10

SilentPotat0, 2020-05-28:
**Describe the bug**
I followed the guide for the automated install and ran the steps for the eclipse setup and it appears everything worked, intill i opened up the template project and it couldnt find any of the the includes

**To Reproduce**
install the esp eclipse plugin following the steps descriped in the readme.md

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Environment Information:** 
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/4697672/eclipse_config.txt)

**ESP-IDF Tools Information:**
doesnt want to tell me anything, nothing happens

kolipakakondal, 2020-05-29:
Hi @SilentPotat0 

- Looks like you don't have the latest version of a plugin - please get the latest using the update site
- Unresolved inclusion errors will disappear only after the build. Is the build successful?
- Have you installed the tools from eclipse? Any issues reported?

SilentPotat0, 2020-05-29:
nvm today it apparently works fine, even tried rebooting yesterday so dont know why. sorry for wasting your time :)

closed 2020-05-29 by SilentPotat0