espressif/idf-eclipse-plugin #93: Installation cannot be completed as requested with CDT 9.10

rherrmannr, 2020-06-04:
**Describe the bug**
I'm trying to install Espressif into a clean Eclipse CDT 2019-12 using CDT 9.10.

**To Reproduce**
Steps to reproduce the behavior:
1. Download Eclipse C/C++ 2019-12 here:
https://www.eclipse.org/downloads/packages/release/2019-12/r
2. unzip, run
3. install new software as described here:
https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md
4. Following error:
```
Cannot complete the install because of a conflicting dependency.
  Software being installed: Espressif IDF Plugins for Eclipse 1.1.1.202006040612 (com.espressif.idf.feature.feature.group 1.1.1.202006040612)
  Software currently installed: C/C++ Qt Support - Preview 9.10.0.201911192140 (org.eclipse.cdt.qt.feature.group 9.10.0.201911192140)
  Only one of the following can be installed at once: 
    Template Engine Freemarker Support (Java 11) 1.1.100.202001140305 (org.eclipse.tools.templates.freemarker.java11 1.1.100.202001140305)
    Template Engine Freemarker Support (Java 11) 1.1.2.201911192140 (org.eclipse.tools.templates.freemarker.java11 1.1.2.201911192140)
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 1.1.1.202006040612 (com.espressif.idf.feature.feature.group 1.1.1.202006040612)
    To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.100.202001140305,1.1.100.202001140305]
  Cannot satisfy dependency:
    From: C/C++ Qt Support - Preview 9.10.0.201911192140 (org.eclipse.cdt.qt.feature.group 9.10.0.201911192140)
    To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.2.201911192140,1.1.2.201911192140]
```

**Expected behavior**
[Installing Prerequisites](https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/) says Eclipse 2018-12 CDT and above are supported.

**Environment Information:** 

[config.txt](https://github.com/espressif/idf-eclipse-plugin/files/4729812/config.txt)

kolipakakondal, 2020-06-05:
Thanks for reporting. I'm looking into it.

kolipakakondal, 2020-06-05:
Hello @rherrmannr 

Until we provide a fix for this issue, you can choose one of these options to proceed with the installation.
1. **Upgrade to Eclipse CDT 2020-03**  - If you've already had Eclipse CDT 2020-03 you don't get into this issue.
2. Choose "**Update my installation to compatible with the items being installed**" and proceed with the installation. This basically will update your Eclipse CDT version to 2020-03

[image]

rherrmannr, 2020-06-05:
Hey @kolipakakondal 

Thanks for your help. I've already seen this workaround, but this won't work for me, because of other dependencies in my eclipse that force me to use CDT 9.10.

Seems like I must wait for your fix or somehow update to CDT 9.11.

thehellmaker, 2020-06-20:
Has the same issue on 2020-06 version of eclipse. Downgraded to 2020-03 it was resolved. Just as FYI

kolipakakondal, 2020-06-23:
We recommend using Java 11 and above while working with IDF Eclipse Pluginconsidering Eclipse 2020-06 has a requirement for Java 11 to work with the CDT

I've added a note on this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.1

groskopf, 2020-08-07:
I had similar problems. I just upgraded my openJdk jre from version 8 to 14. Then it worked.

rherrmannr, 2020-08-10:
I did not test this so far but perhaps this works for others. You can skip the JRE 11 compatibility check:
https://www.eclipse.org/lists/cdt-dev/msg34492.html

kolipakakondal, 2020-08-26:
Hi All,

We have released ESP-IDF Eclipse Plugin v1.2.1 (Build: 1.2.1.202008251732). Please check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.1

Use Java 11 and above, Eclipse 2020-06(4.16) to work with the IDF Eclipse plugin

Thanks

closed 2020-08-26 by kolipakakondal