espressif/idf-eclipse-plugin #96: Multiple projects trying to share a launch configuration

SolidStateLEDLighting, 2020-06-09:
I have 3 sample projects loaded in the workspace -- they individually compiled and ran -- but each new project seems to take over the launch configuration.     At some point, the single launch configuration entry becomes unworkable for 2 out of the 3 projects.    Should loading examples create a new launch configuration for each build?    if not -- does a single launch configuration adapt to each loaded/opened project?

I need to understand the design intent --  Are are launch configurations selectable for any project? 

The build setting page seems to be lost for 2 out of the 3 projects when I revert back to previously built and run projects.

I don't know what I should be expecting...

**To Reproduce**
Steps to reproduce the behavior:
1. Load 3 example projects -- build each in turn and run.
2. Try to return to the previously loaded projects.
3. Launch Configuration is hosed for all but the last project run.
4. Documentation is not clear on the sharing the singularity of the launch configuration --- more explanation please.

**Expected behavior**
I expect all successfully loaded and compiled projects to stay in a buildable form and not break when I load another project.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[Configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/4751078/Configuration.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
[InstalledTools.txt](https://github.com/espressif/idf-eclipse-plugin/files/4751081/InstalledTools.txt)

SolidStateLEDLighting, 2020-06-09:
I have a new observation which is significant.    When I start deleting projects -- previously hidden launch configurations begin to show.... so I think the problem is that we are losing the ability to see those created launch configuration when new projects are loaded....

sigmaaa, 2021-09-21:
Hi @SolidStateLEDLighting,

the fix for this issue is available in the latest [release](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0)

closed 2021-09-21 by sigmaaa