espressif/idf-eclipse-plugin #97: No longer able to create a project from template????

SolidStateLEDLighting, 2020-06-09:
I just deleted 3 sample projects that were all created from templates.... 

When I start fresh again -- the wizard interface won't allow me to "Next"

[image]

Even when I restart Eclispe -- I still can't get back to the templates.   What did do wrong?

SolidStateLEDLighting, 2020-06-09:
I figured out what I needed to do.    I had to go through the steps of installing the esp tools again and setting the path variables as seen in the instructions just before project creation.    That set up the system to allow me again to access the templates for project building...

BUT   BUT -- now I'm getting a  "No Toolchain found for Target Local"

SolidStateLEDLighting, 2020-06-09:
My tool allows me to create a new project with a template -- but it won't build -- and I'm not allowed to edit the launch configuration.

The complaint is "No Toolchain found for Target Local"

I don't even know at this point how to get the entire build environment working again....

SolidStateLEDLighting, 2020-06-09:
NOTE --- when generating a launch target -- the tool doesn't offer me the IDF Target drop-down box.    Looks like it is not aware of my idf project classification.

kolipakakondal, 2020-06-09:
> "No Toolchain found for Target Local"

The issue could be this https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#no-esp-launch-target-found-please-createselect-the-correct-launch-target-error

Make sure you've created esp launch target and selected during the build

SolidStateLEDLighting, 2020-06-09:
The log in the metadata table (project directly) is full of java.lang.NullPointerExceptions

SolidStateLEDLighting, 2020-06-09:
That comment above is interesting -- because there was not launch target at the time.   Normally you work from left to right to complete a sequence and I wasn't at that point.

I did jump ahead and created a target - then went back build and it said that it couldn't find the launch target. (even though I had just created one).

kolipakakondal, 2020-06-09:
> The log in the metadata table (project directly) is full of java.lang.NullPointerExceptions

I think your workspace is corrupted since you've modified the .metadata. I would suggest to create a new workspace and import the existing project

SolidStateLEDLighting, 2020-06-09:
I don't need to import any existing -- I'll just try to get started from scratch again.   I think you much run the "import tools" set when you create a new workspace.

SolidStateLEDLighting, 2020-06-09:
We would be told "Hey -- don't delete the metadata"

SolidStateLEDLighting, 2020-06-09:
I have no idea how to create a new workspace

kolipakakondal, 2020-06-09:
File > Switch Workspace > Other

SolidStateLEDLighting, 2020-06-09:
I see that...  Thank youl.

SolidStateLEDLighting, 2020-06-09:
I have a new workspace.   I asked to install tools again to supply idf path and git.exe path.  

When I started a new project I can't build it... I get "No Toolchain found for Target Local" and the launch configuration will not open in edit mode.     What am I doing wrong?

SolidStateLEDLighting, 2020-06-09:
Forgot to create a launch target.... Created what I think the right launch target - now I get

No esp launch target found. Please create/select the correct 'Launch Target'

kolipakakondal, 2020-06-09:
I'm sorry, did you follow these steps?
https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target

[image]

SolidStateLEDLighting, 2020-06-09:
[image]

SolidStateLEDLighting, 2020-06-09:
It never prompted me with that dialog box.

SolidStateLEDLighting, 2020-06-09:
WAIT -- Ok, I missed that step.     I see now that I much do this manually just after creating the project.   Let me try this again.

SolidStateLEDLighting, 2020-06-09:
It worked this time... boy -- that is a very big stumbling point.....  If you don't follow that exactly -- then you're hosed up big time.

SolidStateLEDLighting, 2020-06-09:
The project is building....

kolipakakondal, 2020-06-09:
See if this helps https://youtu.be/DqnZy4d03_s

Check the demo part

SolidStateLEDLighting, 2020-06-09:
Already watched that video..

SolidStateLEDLighting, 2020-06-09:
I see him doing a lot more in the launch configuration area.... I need to study that again... I may have stopped the vide prior to that area -- or I was so swamped by all the new info that I could not remember what had done there.

SolidStateLEDLighting, 2020-06-09:
My Hello-World project is running.   I just need to gain enough understanding of renaming a project and managing several projects to get this under control.

SolidStateLEDLighting, 2020-06-09:
RECOMMENDATION.... if creating a Launch Target is critical to not breaking the Launch Configuration area -- you should put the Launch Target window to the left of the Configuration window because you can easily forget this operational order.

SolidStateLEDLighting, 2020-06-09:
You helped me a great deal today.     I'm not quite there yet -- but I see the error of my ways in not creating the Launch Target first.   Thank you very much.    I hope you guys can really get this plug-in nailed down because I'd like to stay with Eclipse and see if the tool proves out for ESP development.

closed 2021-02-16 by kolipakakondal