espressif/idf-eclipse-plugin #98: Can't delete a Launch Target

SolidStateLEDLighting, 2020-06-09:
This issue is pretty simple -- there is no intuitive way to delete any ESP32 launch targets.   I have a mess to clean up because made a launch target in an incorrect way -- how do I destroy one?

nasscube, 2020-10-08:
@SolidStateLEDLighting Did you find a solution ? 
Anybody knows how to delete a launch target on Eclipse ?

SolidStateLEDLighting, 2020-10-08:
I dumped your tool and I develop with VS2019 now.

JKSAR, 2021-02-24:
Hi @SolidStateLEDLighting, I found a way.

In ESP Application chose the launch target and click with right button, then select DELETE

[image]

sigmaaa, 2021-08-17:
The option to delete a launch target was added. You can find related changes in [2.2.0 release ](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0). To delete the launch target, just click on the wheel on the launch target that you want to delete and click on the "Delete" button:

[image]

closed 2021-08-17 by sigmaaa