espressif/idf-eclipse-plugin #100: Include file warnings -- did you guys hide features with the plug-in??

SolidStateLEDLighting, 2020-06-12:
My project builds -- but I have a lot of warning about include files that can't be found.

My feedback from the Eclipse people says there should be custom include paths added -- but those entries don't exist anymore in the Project Properties area....

How do we add paths so the C/C++ editor stops complaining about unresolved references???

Working with Windows 2020-03 build and all the latest ESP tools.  Example projects load and build and run -- but I have these pesky include warnings.

SolidStateLEDLighting, 2020-06-13:
Working with Windows 10   Plugin Version: 1.1.1.202006040612

All new install -- all up to date tools.

SolidStateLEDLighting, 2020-06-13:
You can't perfect the magical "just works" solution for the indexer -- then I suggest you publish the solid back-door manual solution for now....    Is it all about the Cmake tool for the manual fix?     I don't know Cmake -- I barely can remember all other tools that I need to know.

SolidStateLEDLighting, 2020-06-13:
FURTHER -- When I most certainly have compiler errors -- they don't register as errors during the building process.    I do get a warning of errors before I flash -- but there is no list of real errors other than the unresolved references.

This tool is pretty much unusable until the unresolved include header files problems are cleared up.

kolipakakondal, 2020-06-16:
Could you make sure the ESP launch target is selected while building a project? This could be one of the reasons. Check this https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#unresolved-inclusion-errors-in-the-editor

To troubleshoot further on this issue, could you please provide the following logs/information.

- **CDT Parser Log:**
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-access-cdt-parser-error-log

- **Environment Info:**
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-provide-eclipse-envrionment-and-plugins-information

- Are you seeing unresolved issues only for imported projects or even if you create a new project?

SolidStateLEDLighting, 2020-06-18:
Thanks for the feedback -- I will examine these leads... stand by.

SolidStateLEDLighting, 2020-06-19:
I really appreciate the help.   All you have asked for is here

Option 1 -- making sure the ESP launch target is selected -- checked that -- not the issue...

Project cleans and builds and all the references I right mouse click to "open declaration" lead to the source file with those unresolved declarations.

Index > rebuild does nothing to dismiss unresolved references.

I have not imported entire projects.  I have only started new projects based on project templates (Hello World) -- and then I am importing work from previous VS work into Eclipse.

[Parser.log](https://github.com/espressif/idf-eclipse-plugin/files/4802640/Parser.log)

[EnvirnInfo.txt](https://github.com/espressif/idf-eclipse-plugin/files/4802661/EnvirnInfo.txt)

kolipakakondal, 2022-02-14:
Hello,

With our recent releases v2.4.0/v2.4.1, we have fixed header unresolved issues, indexer problems, and source code navigation problems.

Please get the latest changes using the update site. Please check our release page for more details.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal