espressif/idf-eclipse-plugin #101: No reaction after selecting Instal Tools or List Installed Tools

wsuchy, 2020-06-21:
**Describe the bug**
No window is being open after clicking on Install Tools from the context menu (Help > ESP-IDF Tools Manager)

**To Reproduce**
Steps to reproduce the behaviour:
1. Install Fedora Linux 32
2. Install Eclipse Version: 2020-03 (4.15) Build id: X20200402-2326
3. Install CDT Version: 9.11.1.202006011430 Build id: 20200601-1605
4. Install ESP-IDF Plugin Version: 1.1.1.202006040612
5. Navigate to Help > ESP-IDF Tools Manager >  Install Tools
**Expected behavior**
A configuration window should be shown

**Environment Information:** 
Available upon request, omitted here due to the privacy concerns

**ESP-IDF Tools Information:**
Doesn't work either: Help > Espressif Tools Manager > List Installed Tools

kolipakakondal, 2020-06-22:
Hi @wsuchy 

Could you please share the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

wsuchy, 2020-06-22:
@kolipakakondal this is the only error I saw in the logs
```
null
org.eclipse.ui
Error
Unhandled event loop exception

java.lang.NoSuchMethodError: java.nio.ByteBuffer.flip()Ljava/nio/ByteBuffer;
	at org.eclipse.ui.internal.console.StreamDecoder.<init>(StreamDecoder.java:42)
	at org.eclipse.ui.console.IOConsoleOutputStream.<init>(IOConsoleOutputStream.java:85)
	at org.eclipse.ui.console.MessageConsoleStream.<init>(MessageConsoleStream.java:63)
	at org.eclipse.ui.console.MessageConsole.newMessageStream(MessageConsole.java:130)
	at com.espressif.idf.ui.update.AbstractToolsHandler.activateIDFConsoleView(AbstractToolsHandler.java:112)
	at com.espressif.idf.ui.update.AbstractToolsHandler.execute(AbstractToolsHandler.java:58)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:97)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5687)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1423)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4955)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4448)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:658)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:557)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:606)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1432)
```

kolipakakondal, 2020-06-22:
It seems to be an issue with a Java version. Could you share the Java version which you're using?

You can try with the latest version of Java(LTS version) 11 and see if It helps.

The plugin supports Java 8 212 and above

kolipakakondal, 2020-06-23:
Please check the notes section here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.1

We recommend using Java 11 and higher due to dependency issues which comes with Eclipse 2020-06

closed 2021-02-16 by kolipakakondal