espressif/idf-eclipse-plugin #102: Dependency issues prevent plugin installation on eclipse

bluebaloon, 2020-06-22:
**Describe the bug**
A clear and concise description of what the bug is.

I tried to install the plugin to the configuration stated below ;

Eclipse version : 2020-06 (4.16.0)
Installed Java components : Java 8 update 251 / Java SE development kit 14.0.1

and got the error stated below ;

Cannot complete the install because some dependencies are not satisfiable
  Software being installed: a.jre.javase 10.0.0
  Software currently installed: C/C++ Autotools support 9.11.1.202004012023 (org.eclipse.cdt.autotools.feature.group 9.11.1.202004012023)
  Cannot satisfy dependency:
    From: C/C++ Autotools support 9.11.1.202004012023 (org.eclipse.cdt.autotools.feature.group 9.11.1.202004012023)
    To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.100.202001140305,1.1.100.202001140305]
  Cannot satisfy dependency:
    From: Template Engine Freemarker Support (Java 11) 1.1.100.202001140305 (org.eclipse.tools.templates.freemarker.java11 1.1.100.202001140305)
    To: osgi.ee; (&(osgi.ee=JavaSE)(version=11))

I'm not a java developer, i just installed the java components to have the plugin work.
What versions of java should i use to overcome this issue. (Please provide link or accurate
version id. There are many types and versions and i can't figure it out.)

Thanks

kolipakakondal, 2020-06-23:
Hi @bluebaloon 

Thanks for reporting. I've just added a note on this issue. Please check this https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.1

Can you please install Java 11 and above to work with this plugin. It seems to be your Eclipse is launched with Java 10

kolipakakondal, 2020-06-23:
You can pick the Java 11 version from here based on the OS https://www.oracle.com/java/technologies/javase-jdk11-downloads.html

bluebaloon, 2020-06-23:
I have removed the eclipse and reinstalled it by selecting jdk 14 from the configuration window. This solved the problem of dependencies. I have installaed the ESP-IDF plugin without any problems and it worked. 

Thanks for help.

Note : You may try to add the lines given below into eclipse.ini file. It may help your current eclipse installation use the new java version.

-vm
C:/Program Files/Java/jdk-14.0.1/bin

closed 2020-06-23 by bluebaloon