espressif/idf-eclipse-plugin #103: Can't compile hello world example with eclipse plugin (fail at compiling stdatomic)

MrXavierB, 2020-07-08:
Hi,

I have difficulties to compile the example code with the ESP-IDF plugin for eclipse. I get several errors at step 365 of 827 in the stdatomic.

I have followed the steps to install all the requirements (using the ESP-IDF Tools setup 2.3 with IDF version 4.0.1) and the plugin. I attached the log file, it seems to be able to compile but I get stuck at this part and not sure what is wrong.

Here is the versions of the tools I have:
Windows 10 (64bit)
Python 3.7 (64bit) - Provided by installer
Git 2.21.0.windows.1 - Provided by installer
IDF version 4.0.1 - Provided by installer
xtensa esp-2019r2-8.2.0 - Provided by installer
cmake 3.13.4 - Provided by installer
openocd-esp32 0.10.0 - Provided by installer
mconf 4.6.0.0 - Provided by installer
ninja 1.9.0 - Provided by installer
ccache 3.7
Eclipse 2019-12 updated to 2020-06 (I could not install the plugin in the 2020-06, I add to use the 2019-12 version first then update).

Thank you!

[log.txt](https://github.com/espressif/idf-eclipse-plugin/files/4888102/log.txt)

kolipakakondal, 2020-07-10:
@igrr @projectgus - Here is the error I could see in the build log, could you guide us on this.

```
C:/Projets/ESP32/ESP_IDF/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':

C:/Projets/ESP32/ESP_IDF/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union

  __atomic_exchange_n(&(object)->__val, desired, order)

                               ^
C:/Projets/ESP32/ESP_IDF/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'

  return (atomic_exchange_explicit(&__object->__flag, 1, __order));

          ^
```

projectgus, 2020-07-10:
Hi @MrXavierB @kolipakakondal ,

That's odd, with gcc 8 compiler this header directory (`components/newlib/include`) shouldn't be included in the build at all (it's included for compatibility with gcc 5). However the other compiler command line arguments indicate that gcc 8 is being used, as expected.

@MrXavierB  Can you please try deleting the entire build directory (ie `C:\Projets\ESP32\workspace2019_12_02\Test_Hello_World\build`) and then build the project in Eclipse again? This should not ever be necessary, but it may be because of some bug that CMake hasn't regenerated some of the build files.

Please capture the full log from this build and post it if it fails again, it will show the output from when CMake is analyzing the toolchain and the project requirements.

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release.

closed 2021-02-16 by kolipakakondal