espressif/idf-eclipse-plugin #104: Not able to Build with Ninja Error

phy455, 2020-07-08:
Hello,

I have followed instllation guide and created created new project as mentioned in  [https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#create-a-new-project](url)

I am not able to build it with the following error 

```
Building in: E:\Workspace\idf_test\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=E:\IDF\esp-idf-v4.0\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 E:\Workspace\idf_test
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "C:\\Users\\pratik.yadav.455\\AppData\\Local\\Programs\\Python\\Python37\\python.exe -m pip install --user -r E:\\IDF\\esp-idf-v4.0\\requirements.txt" for resolving the issue.
CMake Error at E:/IDF/esp-idf-v4.0/tools/cmake/build.cmake:270 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  E:/IDF/esp-idf-v4.0/tools/cmake/build.cmake:396 (__build_check_python)
  E:/IDF/esp-idf-v4.0/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Workspace/idf_test/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.

Build complete (0 errors, 0 warnings): E:\Workspace\idf_test\build
```

I have run list install tools and its log is as mentioned below.

```
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
```

Is there any other configuration required?

phy455, 2020-07-08:
My installation logs are as following 
```
Installing tools...
This can take a while. Please be patient.
Executing C:\Users\pratik.yadav.455\AppData\Local\Programs\Python\Python37\python.exe E:\IDF\esp-idf-v4.0\tools\idf_tools.py install all
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)

Executing C:\Users\pratik.yadav.455\AppData\Local\Programs\Python\Python37\python.exe E:\IDF\esp-idf-v4.0\tools\idf_tools.py install-python-env
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                  [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                  [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Creating a new Python environment in E:\IDF\.espressif\python_env\idf4.0_py3.7_env
Traceback (most recent call last):
  File "E:\IDF\esp-idf-v4.0\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "E:\IDF\esp-idf-v4.0\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "E:\IDF\esp-idf-v4.0\tools\idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\pratik.yadav.455\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\pratik.yadav.455\\AppData\\Local\\Programs\\Python\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'E:\\IDF\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 2.

Executing C:\Users\pratik.yadav.455\AppData\Local\Programs\Python\Python37\python.exe E:\IDF\esp-idf-v4.0\tools\idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=E:\IDF\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE=1
PATH=E:\IDF\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin;E:\IDF\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;E:\IDF\.espressif\tools\cmake\3.13.4\bin;E:\IDF\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin;E:\IDF\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;E:\IDF\.espressif\tools\ninja\1.9.0\;E:\IDF\.espressif\tools\idf-exe\1.0.1\;E:\IDF\.espressif\tools\ccache\3.7\;E:\IDF\esp-idf-v4.0\tools;%PATH%
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.
```

phy455, 2020-07-08:
Error was due to given steps don't run requirement.txt file.

Manually run requirment.txt file which solve issue in eclipse terminal.

Please update the installation step or change installation script.

kolipakakondal, 2020-07-10:
> Error was due to given steps don't run requirement.txt file.
> 
> Manually run requirment.txt file in eclipse terminal which solve issue.
> 
> Please update the installation step or change installation script.

@igrr - Shall we run **requirements.txt** command before we install the tools in the eclipse plugin? 
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started-cmake/index.html#step-4-install-the-required-python-packages

`python -m pip install --user -r $IDF_PATH/requirements.txt`

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release.

closed 2021-02-16 by kolipakakondal