espressif/idf-eclipse-plugin #105: sdkconfig SPI RAM config are not available

ftoyon, 2020-07-15:
In 'SDK Configuration' page when we check 'Support for external, SPI-connected RAM' in 'ES32-Specific' the 'SPI RAM config' are not available.

[image]
[image]

IDF Eclipse Plugin Version: 1.1.1.202006040612

IDF Version: 4.0.1

kolipakakondal, 2020-07-16:
Thanks for reporting, let us check.

kolipakakondal, 2020-08-26:
Hi All,

We have fixed this issue with our recent releases, Please check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.1

Use Java 11 and above, Eclipse 2020-06(4.16) to work with the IDF Eclipse plugin

Thanks

closed 2020-08-26 by kolipakakondal

ftoyon, 2020-08-26:
Hi !

After update plugin to the version 1.2.1
We have this error
[image]

And in Json Configuration Server Console:
```
The following Python requirements are not satisfied:

esp-windows-curses; sys_platform == 'win32'

To install the missing packages, please run "C:\Users\Frederic\esp32\esp-idf-v4.1\install.bat"

ESP-IDF v4.1
```
I'm under Windows with eclipse 2020-06 and java 11.0.7 and phyton 3.8.

I get the esp-idf 4.1yesterday by git clone and after I run install.bat and export.bat

The `idf.py menuconfig` is working.