espressif/idf-eclipse-plugin #106: ESP-IDF Eclipse Plugin installation failed

gilles380, 2020-07-23:
Hi,

with Prerequisites :
java version "1.8.0_231"
Java(TM) SE Runtime Environment (build 1.8.0_231-b11)
and ECLIPSE IDE :
Version: 2020-06 (4.16.0)
Build id: 20200615-1200

The following plugins failed to install => please note that PLUGINS VERSIONS are different from the one pictured in the git hub howto

Software currently installed: C/C++ Autotools support 9.11.1.202004012023 (org.eclipse.cdt.autotools.feature.group 9.11.1.202004012023)
Cannot satisfy dependency:
From: C/C++ Autotools support 9.11.1.202004012023 (org.eclipse.cdt.autotools.feature.group 9.11.1.202004012023)
To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.100.202001140305,1.1.100.202001140305]
Cannot satisfy dependency:
From: Template Engine Freemarker Support (Java 11) 1.1.100.202001140305 (org.eclipse.tools.templates.freemarker.java11 1.1.100.202001140305)
To: osgi.ee; (&(osgi.ee=JavaSE)(version=11))

[image]

kolipakakondal, 2020-07-23:
Hi @gilles380 

That's because of Java 8. Could you please update the java version to Java 11 and above. Sorry for not updating this in the documentation. I've added a note on this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.1

kolipakakondal, 2020-08-26:
Hi All,

We have released ESP-IDF Eclipse Plugin v1.2.1 (Build: 1.2.1.202008251732). Please check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.1

Use Java 11 and above, Eclipse 2020-06(4.16) to work with the IDF Eclipse plugin

Thanks

closed 2020-08-26 by kolipakakondal