espressif/idf-eclipse-plugin #107: Include Search Path not working for components in a project

bittailor, 2020-07-23:
**Describe the bug**
The c++ indexer does not properly work for c/c++ files in a component in a project. I think the problem is the "Include Search Path" for the files in a component.

When I create a CDT parser log file (Project > C/C++ Index > Create Parser Log) for a file in a component the ESP-IDF paths are not present

```
File:                  file:/---/bt-core/components/BtCoreEspIdf/src/Bt/Storage/NvsRepository.h
Language:              GNU C++
Index Version:         215.0
Build Configuration:   Configuration
Context:               file:/---/bt-core/main/BtAlarmClockApp.cpp
   C++, {}
Versions in Index:     1
   C++: {}; 1 macros, 2 includes, 20 names;

Include Search Path (option -I):
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0/xtensa-esp32-elf
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0/backward
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include-fixed
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/sys-include
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include

Macro definitions (option -D):
...
```

When I create a CDT parser log file for a file in the main folder of a project the ESP-IDF paths are there:

```
File:                     file:/---/bt-core/main/BtAlarmClockApp.cpp
Language:              GNU C++
Index Version:         215.0
Build Configuration:   Configuration
Context:               file:/---/bt-core/main/BtAlarmClockApp.cpp
   C++, {}
Versions in Index:     1
   C++: {}; 0 macros, 13 includes, 762 names;

Include Search Path (option -I):
   /Users/bittailor/esp/esp-idf/components/newlib/platform_include
   /Users/bittailor/esp/esp-idf/components/freertos/include
   /Users/bittailor/esp/esp-idf/components/freertos/xtensa/include
   /Users/bittailor/esp/esp-idf/components/heap/include
   /Users/bittailor/esp/esp-idf/components/log/include
   /Users/bittailor/esp/esp-idf/components/lwip/include/apps
   /Users/bittailor/esp/esp-idf/components/lwip/include/apps/sntp
   /Users/bittailor/esp/esp-idf/components/lwip/lwip/src/include
   /Users/bittailor/esp/esp-idf/components/lwip/port/esp32/include
   /Users/bittailor/esp/esp-idf/components/lwip/port/esp32/include/arch
   /Users/bittailor/esp/esp-idf/components/soc/src/esp32/.
   /Users/bittailor/esp/esp-idf/components/soc/src/esp32/include
   /Users/bittailor/esp/esp-idf/components/soc/include
   /Users/bittailor/esp/esp-idf/components/esp_rom/include
   /Users/bittailor/esp/esp-idf/components/esp_common/include
   /Users/bittailor/esp/esp-idf/components/esp_system/include
   /Users/bittailor/esp/esp-idf/components/xtensa/include
   /Users/bittailor/esp/esp-idf/components/xtensa/esp32/include
   /Users/bittailor/esp/esp-idf/components/esp32/include
   /Users/bittailor/esp/esp-idf/components/driver/include
   /Users/bittailor/esp/esp-idf/components/driver/esp32/include
   /Users/bittailor/esp/esp-idf/components/esp_ringbuf/include
   /Users/bittailor/esp/esp-idf/components/efuse/include
   /Users/bittailor/esp/esp-idf/components/efuse/esp32/include
   /Users/bittailor/esp/esp-idf/components/espcoredump/include
   /Users/bittailor/esp/esp-idf/components/esp_timer/include
   /Users/bittailor/esp/esp-idf/components/soc/soc/esp32/.
   /Users/bittailor/esp/esp-idf/components/soc/soc/esp32/include
   /Users/bittailor/esp/esp-idf/components/soc/soc/esp32/../include
   /Users/bittailor/esp/esp-idf/components/vfs/include
   /Users/bittailor/esp/esp-idf/components/esp_wifi/include
   /Users/bittailor/esp/esp-idf/components/esp_wifi/esp32/include
   /Users/bittailor/esp/esp-idf/components/esp_event/include
   /Users/bittailor/esp/esp-idf/components/esp_netif/include
   /Users/bittailor/esp/esp-idf/components/esp_eth/include
   /Users/bittailor/esp/esp-idf/components/tcpip_adapter/include
   /Users/bittailor/esp/esp-idf/components/app_trace/include
   /Users/bittailor/Documents/Code/Esp32/bt-core/components/BtCore/src
   /Users/bittailor/Documents/Code/Esp32/bt-core/components/Date
   /Users/bittailor/Documents/Code/Esp32/bt-core/components/BtCoreEspIdf/src
   /Users/bittailor/esp/esp-idf/components/nvs_flash/include
   /Users/bittailor/esp/esp-idf/components/spi_flash/include
   /Users/bittailor/esp/esp-idf/components/mbedtls/port/include
   /Users/bittailor/esp/esp-idf/components/mbedtls/mbedtls/include
   /Users/bittailor/esp/esp-idf/components/mbedtls/esp_crt_bundle/include
   /Users/bittailor/Documents/Code/Esp32/bt-core/components/BtAlarmClock/src
   /Users/bittailor/Documents/Code/Esp32/bt-core/components/ArduinoJson
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0/xtensa-esp32-elf
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include/c++/8.2.0/backward
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include-fixed
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/sys-include
   /Users/bittailor/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/include

Macro definitions (option -D):
...

```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a component in a project (e.g.: <project>/components/MyComponent
2. Create a c/c++ file in that component and include a ESP-IDF header (e.g.: #include <nvs_flash.h>)
3. Add the REQUIRES to the components CMakeLists.txt (e.g.: REQUIRES nvs_flash)  
3. Build the project.
4. The project with the component builds without error but the indexer complains unresolved inclusion for the  <nvs_flash.h> in the components c/c++ file.
5. The same header added to a  c/c++ file in the main folder of the project works also for the indexer.

**Expected behavior**
c++ indexer also works for the files in a component in a project.

**Screenshots**
[image]
[image]

**Environment Information:** 
[EnvironmentInformation.txt](https://github.com/espressif/idf-eclipse-plugin/files/4968962/EnvironmentInformation.txt)

**ESP-IDF Tools Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200420 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	1.1.1.202006040612	com.espressif.idf.feature.feature.group

kolipakakondal, 2020-07-24:
Thanks for the report. Let us look into it.

kolipakakondal, 2020-07-27:
Hi @bittailor 

I tried this example where it has components in the project and it seems to work fine for me. Could you check this for reference for defining CMakeList.txt and others

https://github.com/espressif/esp-idf/tree/release/v4.0/examples/build_system/cmake/import_prebuilt

kolipakakondal, 2020-07-27:
[hello_world.zip](https://github.com/espressif/idf-eclipse-plugin/files/4980769/hello_world.zip)

Here is the another example which I created based on the scenario which you've explained but somehow I'm not able to reproduce the issue.

[image]

bittailor, 2020-07-28:
Hi @kolipakakondal 

Thanks for looking into this. Your hello_world.zip example works and finds the headers 🤔. I will try to find the difference to my project and also to create a _small_ example project that tries to reproduce my issue.

bittailor, 2020-07-29:
Hi @kolipakakondal 

I was now able to create a _small_ example that reproduces my issue and with this also found what the difference that creates the issue. It is a consequence that I nested projects inside a project. 

So I think from my point of view we can close this issue as it is a consequence of nesting my _app projects_ inside my _components project_, which might not be intended and is simple to change/avoid.  

Here now the details to reproduce the issue if you are interested:

You find the example project in the [bittailor/IssueDemo-IdfEclipsePlugin-IEP-214](https://github.com/bittailor/IssueDemo-IdfEclipsePlugin-IEP-214) repository.  When you import the project into eclipse please select the "Search for nested projects" and import all three projects.

[image]

Now when you built all the three projects the "include issue" is present in the `/demo-indexer-issue/components/CmpOne/src/Cmp/One/ClassB.h` file.

[image]
 
I think the issue is a consequence of the nested `DemoIndexerIssueApp` project in the apps folder under the `demo-indexer-issue` project.  I created this inspired by the test project in the [unit_test example](https://github.com/espressif/esp-idf/tree/master/examples/system/unit_test/test), but it seems not to be a good idea. When I move the `DemoIndexerIssueApp` project outside the `demo-indexer-issue` project (so no nesting) the indexer issue is gone.

For the issue to be preset the `DemoIndexerIssueApp` needs to be nested and must include the `Cmp/One/ClassB.h` in one of it cpp files. Then the Parser Log file for ClassB.h is wrong and an interesting difference is that it states `DemoIndexerIssueApp.cpp` as its context and does not include the needed _Include Search Paths_

```
Context:               file:/.../demo-indexer-issue/apps/DemoIndexerIssueApp/main/DemoIndexerIssueApp.cpp
```

When I do move the `DemoIndexerIssueApp` out of the `demo-indexer-issue` project (or already if I do not include `Cmp/One/ClassB.h` in one of it cpp files inside `DemoIndexerIssueApp`) the issue is gone Parser Log file states `ClassB.cpp` as context and contains the needed _Include Search Paths_

```
Context:               file:/.../demo-indexer-issue/components/CmpOne/src/Cmp/One/ClassB.cpp
```

closed 2020-08-13 by bittailor