espressif/idf-eclipse-plugin #108: can't download the elf file to the target when running sample prov_dev_client_ll_sample

Mode79, 2020-07-26:
**Describe the bug**
i can see the two bin files for the "bootloader.elf" and "prov_dev_client_ll_sample.elf" but when i try to run so i can download the code, it gives me the following error

Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.27.0.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/mfawz/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/mfawz/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/mfawz/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/mfawz/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/mfawz/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Loading defaults file E:/esp-azure/examples/prov_dev_client_ll_sample/sdkconfig.defaults...
C:\Users\mfawz\AppData\Local\Temp\confgen_tmpkrmpbqnq:5 line was updated to CONFIG_NEWLIB_NANO_FORMAT=n
-- Found PythonInterp: C:/Users/mfawz/.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe (found version "3.8.3") 
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "prov_dev_client_ll_sample" version: v1.0-23-gf6bd5cd-dirty
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script E:/esp-azure/examples/prov_dev_client_ll_sample/build/default/esp-idf/esp32/esp32_out.ld
-- Adding linker script E:/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script E:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: E:/esp-idf/components/app_trace E:/esp-idf/components/app_update E:/esp-idf/components/asio E:/esp-idf/components/bootloader E:/esp-idf/components/bootloader_support E:/esp-idf/components/bt E:/esp-idf/components/cbor E:/esp-idf/components/coap E:/esp-idf/components/console E:/esp-idf/components/cxx E:/esp-idf/components/driver E:/esp-idf/components/efuse E:/esp-idf/components/esp-tls E:/esp-idf/components/esp32 E:/esp-idf/components/esp_adc_cal E:/esp-idf/components/esp_common E:/esp-idf/components/esp_eth E:/esp-idf/components/esp_event E:/esp-idf/components/esp_gdbstub E:/esp-idf/components/esp_hid E:/esp-idf/components/esp_http_client E:/esp-idf/components/esp_http_server E:/esp-idf/components/esp_https_ota E:/esp-idf/components/esp_https_server E:/esp-idf/components/esp_local_ctrl E:/esp-idf/components/esp_netif E:/esp-idf/components/esp_ringbuf E:/esp-idf/components/esp_rom E:/esp-idf/components/esp_serial_slave_link E:/esp-idf/components/esp_system E:/esp-idf/components/esp_timer E:/esp-idf/components/esp_websocket_client E:/esp-idf/components/esp_wifi E:/esp-idf/components/espcoredump E:/esp-idf/components/esptool_py E:/esp-idf/components/expat E:/esp-idf/components/fatfs E:/esp-idf/components/freemodbus E:/esp-idf/components/freertos E:/esp-idf/components/heap E:/esp-idf/components/idf_test E:/esp-idf/components/jsmn E:/esp-idf/components/json E:/esp-idf/components/libsodium E:/esp-idf/components/log E:/esp-idf/components/lwip E:/esp-azure/examples/prov_dev_client_ll_sample/main E:/esp-idf/components/mbedtls E:/esp-idf/components/mdns E:/esp-idf/components/mqtt E:/esp-idf/components/newlib E:/esp-idf/components/nghttp E:/esp-idf/components/nvs_flash E:/esp-idf/components/openssl E:/esp-idf/components/partition_table E:/esp-idf/components/perfmon E:/esp-idf/components/protobuf-c E:/esp-idf/components/protocomm E:/esp-idf/components/pthread E:/esp-idf/components/sdmmc E:/esp-idf/components/soc E:/esp-idf/components/spi_flash E:/esp-idf/components/spiffs E:/esp-idf/components/tcp_transport E:/esp-idf/components/tcpip_adapter E:/esp-idf/components/tinyusb E:/esp-idf/components/ulp E:/esp-idf/components/unity E:/esp-idf/components/vfs E:/esp-idf/components/wear_levelling E:/esp-idf/components/wifi_provisioning E:/esp-idf/components/wpa_supplicant E:/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: E:/esp-azure/examples/prov_dev_client_ll_sample/build/default
[1/947] Generating project_elf_src.c
[2/947] Generating linker script...
[3/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cmac.c.obj
[4/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/blowfish.c.obj
[5/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chacha20.c.obj
[6/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chachapoly.c.obj
[7/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1write.c.obj
[8/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj
[9/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cipher_wrap.c.obj
[10/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/cipher.c.obj
[11/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/base64.c.obj
[12/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/camellia.c.obj
[13/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/des.c.obj
[14/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1parse.c.obj
[15/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum.c.obj
[16/947] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[17/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ccm.c.obj
[18/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecdh.c.obj
[19/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ctr_drbg.c.obj
[20/947] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/dhm.c.obj
[... 509 lines cut ...]
**ESP-IDF Tools Information:**
Executing C:\Users\mfawz\AppData\Local\Programs\Python\Python38-32\python.exe E:\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r1-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
[... cut ...]
  - v0.10.0-esp32-20200420 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	1.1.1.202006040612	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

mpapini, 2020-09-16:
I had the same issue, but moving the path to python (in my case C:\Users\user\AppData\Local\Programs\Python\Python38\) to the beginning of the path entered in Preferences -> C/C++ -> Build -> PATH worked for me.

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent releases.

closed 2021-02-16 by kolipakakondal