espressif/idf-eclipse-plugin #109: Eclipse autocomplete does not work

sirgrej, 2020-08-11:
**Describe the bug**
Eclipse Autocomplete (ctrl. + space) does not work when using espressif plugin. It works for standart gcc libs but not for esp-idf.

- Building +flashing works fine 
- no unresolved inclusion
- no Errors

**To Reproduce**
try to use auto-completion by typing (ctrl. + Space) while coding

**Expected behavior**
Normally I would just add the esp-idf include Path to the Path and Symbols Tab in Project Properties, but this tab is missing in newer eclipse versions

**Environment Information:** 
Eclipse 2020-06
java JDK 11

**ESP-IDF Tools Information:**

* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2019r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

**IDF Eclipse Plugin Version:** 
Espressif plugin 1.1.1.202006040612

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

kolipakakondal, 2020-08-12:
[image]
[image]
[image]

Hi @sirgrej 

Could you reset the above preference pages using "**Restore defaults**" and see if it helps?

Thanks

sirgrej, 2020-08-12:
Hi Kondal, thx for your reply,

I have a fresh eclipse install, so restoring to defaults did not help as expected, but checking the "**Parsing-based Proposal**" option in **Editor ->Content Assist -> Advanced** solved to Problem

[image]

This is strange, I guess something went wrong with the eclipse default settings

Any chance to check these setting while installing the espressif plugin?

kolipakakondal, 2020-08-19:
By default "Parsing-based Proposal" is selected in the default content assist in Eclipse.

closed 2021-02-16 by kolipakakondal