espressif/idf-eclipse-plugin #111: Espressif IDF plugin won't install on Eclipse

huzzziiii, 2020-08-17:
I'm following the steps [from here](https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url) to install the plugin in Eclipse but upon doing so, I am getting an error saying It won't install it, as shown [here](https://imgur.com/a/B9MV3Oc).

I usually use STM32CubeIDE but I tried on Eclipse to no avail.

kolipakakondal, 2020-08-19:
Hi @huzzziiii 

You need to have Eclipse 2020-06(4.16) and Java 11 to work the IDF Eclipse Plugin.
Please check the release notes here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.1.1

Thanks

kolipakakondal, 2020-08-26:
Please check our recent release https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.1

Use Java 11 and above, and Eclipse 2020-06(4.16) while working with the IDF Eclipse plugin.

closed 2020-08-26 by kolipakakondal