espressif/idf-eclipse-plugin #112: Getting loader constraint violation upon creating a Espressif IDF project in Eclipse

huzzziiii, 2020-08-17:
I upgraded my Eclipse version to 2020-06 (4.16.0), and been following the steps [from here to create a project](https://github.com/espressif/idf-eclipse-plugin#create-a-new-project) but doing so results in the following error:

`Internal Error: 
loader constraint violation: loader org.eclipse.osgi.internal.loader.EquinoxClassLoader @d76a6bd wants to load abstract class javax.xml.bind.JAXBContext. A different abstract class with the same name was previously loaded by org.eclipse.osgi.internal.loader.EquinoxClassLoader @12d5f24d. (javax.xml.bind.JAXBContext is in unnamed module of loader org.eclipse.osgi.internal.loader.EquinoxClassLoader @12d5f24d, parent loader 'platform')`

kolipakakondal, 2020-08-19:
Hi @huzzziiii 

This seems to be an issue with your Java version.  Please update to the Java 11 and higher.

Thanks

kolipakakondal, 2020-09-07:
We have a new release v1.2.2, please verify and let us know

This requires Eclipse 2020-06 and Java 11 and above.

https://github.com/espressif/idf-eclipse-plugin/issues/114

closed 2020-09-07 by kolipakakondal