espressif/idf-eclipse-plugin #113: Can't get to function definitions with Eclipse with ESP-IDF

freeride01, 2020-08-21:
Hi,
I am using Eclipse with the ESP-IDF plugin. Projects build and run fine, but I can't get to some function definitions, if they are not in a file in /main directory. Ctrl+click on the function gets me at most to the declarations and function prototypes.
For example, in the "bleprph" example project, function `nvs_flash_init()`. Ctrl+click on it takes me to the declaration in nvs_flash.h, then another Ctrl+click doesn't take me anywhere.
Also when I search in the entire project (Ctrl+H -> File search -> Enclosing project), I can't find the definitions.
I am new to Eclipse and have only used it with the ESP-IDF plugin, so I can't say if this happens with other plugins/frameworks as well.
Any help will be much appreciated.

kolipakakondal, 2020-08-24:
Thanks @freeride01 for the feedback. I'll take a look at this.

Can you give what kind of functions which are showing errors? Is it core C functions or esp-idf functions?

freeride01, 2020-08-24:
Yeah. Interestingly, the Console view shows no errors in build, but the Problems view shows hundreds of problems and if I try to upload the project, I get a dialog box that says "Errors exist in required projects". In Problems view, when I double-click on a problem, I get this:

[image]

When I Ctrl+click on "handle", I get this window:

[image]

(I've erased the project name for privacy purposes).

kolipakakondal, 2021-02-16:
Closing as duplicate

closed 2021-02-16 by kolipakakondal