espressif/idf-eclipse-plugin #114: Java version information confusing and update install fails!!!

ammaree, 2020-08-24:
I am using a Macbook Pro with latest Catalina installed
Eclipse is latest version 2020-06 (4.16.0) Build id: 20200615-1200
[image]

System Preferences - Java - Java Control Panel - General - About shows V8U261 (build 1.8.0_261-b12)
[image]

%java --version shows 'java 14.0.1 2020-04-14 (build 14.0.1-7)'
[image]

Eclipse shows updates being available:
[image]

But the update fails with error info being:
[image]

Q1. What version of Java am I running?
Q2. If there is a newer version of java required and available, what is the version and where do I get it?
Q3. How do I fix the update install error so I can get OpenOCD debugging going?

Thanks

ammaree, 2020-08-24:
I think I have resolved Q1 and Q2.

A1. I had both SE Runtime 8u261 and JDK 14.0.1 installed. I have now uninstalled SE Runtime 8u261 so the Control panel is not showing any more.

A2. I have downloaded and installed the minor update from 14.0.1 to 14.0.2

But the primary problem remain, how to get  OpenOCD installed and working. I have tried to uninstall and reinstall the Eclipse ESP IDF Plugin but now the install fails completely.

How can I completely remove all traces of the ESP IDF Plugin and reinstall from scratch?

kolipakakondal, 2020-08-26:
Hi @ammaree 

We have fixed the OpenOCD issue with our latest release, please check this here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.1

Please confirm once it's verified.

Thank you

ammaree, 2020-08-26:
@kolipakakondal 
It is still not working, although the symptoms have changed.

I would like to reopen the issue and supply full details. How do I reopen?

kolipakakondal, 2020-08-26:
@ammaree Could you please provide the error details?

ammaree, 2020-08-26:
Basically started a completely clean installation, deleted Eclipse (application, .project, .cproject, .metadata/* etc) as well as /Users/andremaree/.eclipse and /Users/andremaree/.p2 directories

From command line, checked that compile works using idf.py build, all works with no errors.
Following instruction on Github installed all requisites:
* Single version of java 14.0.2
* Python 3.8 installed, although % python --version shows 2.7.16
* 2020-06 Eclipse, starts clean and recreates ~/.eclipse and ~/.p2 directories.
* git 2.28.0
* ESP-IDF 4.x master at ~/Dropbox/devs/ws/z-sdk/esp-idf

Installed plugin from the update site, no problem.
After eclipse restart, select Help -> ESP-IDF Tools Manager -> Install tools

Option 1:
---------
[image]

Result 1:
--------
[image]

Option 2:
---------
[image]

Result 2:
---------
[image]

Info on Python:
---------------
[image]

ammaree, 2020-08-26:
Just following a hunch I tried the menu option Eclipse -> Help -> Download & Configure ESP IDF.
Step 1:
[image]

Step 2:
[image]

Step 3:
[image]

Step 4:
[image]

What concerns me here is still using Python 2.7 although support will be dropped soon.

kolipakakondal, 2020-08-26:
> OSError: [Errno 2] No such file or directory

This is the error that I could see from the console which is causing an issue while running i**df_tools.py install-python-env** command internally

This is could be because git was not part of the environment path. If not, can you set git to the environment path?

Can you run the below command from the idf path?
$ git describe --tags

ammaree, 2020-08-26:
v4.3-dev-775-ge664a6b4c9

kolipakakondal, 2020-08-26:
> ## Result 1:
> [image]
> 

@projectgus  @igrr  Can you please help with this error?

kolipakakondal, 2020-08-26:
Okay good, now you're able to install tools and proceed further right?

Let me check with the team on the Python version

ammaree, 2020-08-26:
Here is the Python version error:
[image]

ammaree, 2020-08-26:
A step further, next problem but not Python related. 

During the Debug As configuration, 1 field is not automatically populated.
[image]

Since I build firmware for a number of different ESP32 based devices, the name filled in is only correct for 1 model. In order to automate this, what variables should I use to automatically pick up the name from the CMake process?

Since my ESP-WROVER-KIT could be either the 1.8V or 3.3V version, how do I identify the correct version?
[image]

In case it helps with debugging, the 3rd config screen:
[image]

Lastly, when I start the debug process, the following error occurs:
[image]

And the Console output related to the error is:
[image]

Since it detects the processors and the dual cores correctly, it seems likely a small configuration/timing problem?

ammaree, 2020-08-26:
@kolipakakondal 

Took the case off my ESP-WROVER-KIT and verified the module fitted. It is a 1.8V ESP32-WROVER (ESP32-D0WDQ6) so the selected board is correct.

curiousmuch, 2020-08-31:
> A step further, next problem but not Python related.
>
> During the Debug As configuration, 1 field is not automatically populated.
> [image]
>
> Since I build firmware for a number of different ESP32 based [...]

I'm seeing the same issue for debugging.

kolipakakondal, 2020-09-02:
Hi @ammaree @curiousmuch 

Could you please try this local build and confirm https://github.com/espressif/idf-eclipse-plugin/issues/118#issuecomment-685305247

kolipakakondal, 2020-09-07:
JTAG device loading issue got fixed with the v1.2.2

https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.2

closed 2020-09-07 by kolipakakondal