espressif/idf-eclipse-plugin #115: Workspace location can't include space to create ESP-IDF project

zeebak, 2020-08-25:
I am unable to create an ESP32 New Project. 

[image]

freeride01, 2020-08-25:
The path to your workspace (F:\Eclipse Esp\Workspace) contains a space. Try without it - F:\Eclipse_Esp\Workspace for example.

kolipakakondal, 2020-08-26:
Hi @zeebak 

IDF project name or workspace name with spaces are not allowed.

TiSiAi, 2020-09-07:
Hello
On macOS Catalina 10.15.6
I have the same problem, but I found the problem linked with the default Eclipse folder project.
My default project workspace has a space on his name and if change change the path without space when I try to create a new est project it is impossible.
I I launch eclipse with a default project workspace without space, it is ok !
[image]
[image]
[image]
[image]

kolipakakondal, 2020-09-09:
> [image]
This is a bug. Thanks for providing the details. We will fix this in the next release.

closed 2020-09-25 by kolipakakondal