espressif/idf-eclipse-plugin #117: ESP-IDF: Application Size Analysis Failes

curiousmuch, 2020-08-30:
**Describe the bug**
The Application Size Analysis Tools fails. The attached log is generated. 

[Eclipse_Information_20200830.txt](https://github.com/espressif/idf-eclipse-plugin/files/5147540/Eclipse_Information_20200830.txt)

**To Reproduce**
1. Clone ESP32-IDF
2. Checkout v4.1 
3. Install Tools 
4. Install Eclipse 2020-06 
5. Install Espressif Pluging 
6. Import hello_world example and build 
7. Right click on the project and select Application Size Analysis 

**Expected behavior**
What is seen for the ESP-IDF: Application Size Analysis feature in the README. 

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

[Eclipse_Information_20200830.txt](https://github.com/espressif/idf-eclipse-plugin/files/5147538/Eclipse_Information_20200830.txt)

**ESP-IDF Tools Information:**
[image]

**IDF Eclipse Plugin Version:** 
[image]

curiousmuch, 2020-08-31:
Please let me know if there is any additional information I can provide.

kolipakakondal, 2020-09-02:
Hi @curiousmuch

When you encounter an error, please provide the error log as mentioned here https://github.com/espressif/idf-eclipse-plugin#error-log

curiousmuch, 2020-09-03:
Attached Sir! 
[Application-Size.log](https://github.com/espressif/idf-eclipse-plugin/files/5166201/Application-Size.log)

kolipakakondal, 2020-09-03:
Hi @curiousmuch 

What's the IDF version which you're using? I could see the below error in the error log.

```
MESSAGE usage: idf_size - a tool to print IDF elf file sizes [-h]
                                                     [--toolchain-prefix TOOLCHAIN_PREFIX]
                                                     [--json] [--archives]
                                                     [--archive_details ARCHIVE_DETAILS]
                                                     [--files]
                                                     [-o OUTPUT_FILE]
                                                     map_file
idf_size - a tool to print IDF elf file sizes: error: unrecognized arguments: --target esp32
```

closed 2020-09-03 by curiousmuch