espressif/idf-eclipse-plugin #118: OpenOCD / GDB Debugging Plugin Functionality Fails

curiousmuch, 2020-08-30:
**Describe the bug**
I'm able to start a Debugging Session using the command line options (setup GDB Server and connect Client), but not via Eclipse. Further, even if I start the GDB server in the command line and then attempt to attach using the plug-in functionality, it generates the same error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone ESP-IDF 
2. Checkout v4.1  
3. Install all tools 
4. Install Eclipse 202006 and Espressif Plugin 
5. Import hello_world example and Build
6. Add Debugging Setup 
7. Click Debug :) 

**Expected behavior**
It appears from the plug-in README, it will auto setup a GDB server and connect eclipse to it? Please let me know if this is incorrect?

**Screenshots**
Working w/ command line. 
[image]
[image]

Failure w/ Eclipse Plugin (local GBD Server) 
[image]
[image]

Failure w/ Eclipse Plugin (Remote GBD Sever) 
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[Eclipse_Information_20200830.txt](https://github.com/espressif/idf-eclipse-plugin/files/5147571/Eclipse_Information_20200830.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
[image]

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

[image]

curiousmuch, 2020-08-31:
Please let me know if there any additional information I can help provide!

kolipakakondal, 2020-09-01:
Hi @curiousmuch 

Could you please provide the error log as mentioned here https://github.com/espressif/idf-eclipse-plugin#error-log

curiousmuch, 2020-09-01:
Absolutely! I'll be able to upload in a few hours. Thanks for the quick response!

curiousmuch, 2020-09-02:
Sorry for the delay. 
[JTAG-log2.log](https://github.com/espressif/idf-eclipse-plugin/files/5160077/JTAG-log2.log)
[JTAG-log.log](https://github.com/espressif/idf-eclipse-plugin/files/5160078/JTAG-log.log)

I believe these are the logs you want. I wasn't able to get to the office today, but these are the errors setting in the log from previously.

kolipakakondal, 2020-09-02:
Thanks @curiousmuch for providing the logs.

Could you try this local build and confirm it. Based on that, I'll provide release update.

[com.espressif.idf.update-v1.2.1-1-g20abb88e.zip](https://github.com/espressif/idf-eclipse-plugin/files/5160235/com.espressif.idf.update-v1.2.1-1-g20abb88e.zip)

Please follow the steps from here to install the above build https://github.com/espressif/idf-eclipse-plugin#installing-idf-eclipse-plugin-from-local-archive

ammaree, 2020-09-02:
I have installed the local build and although we have progress, based on the time taken till it halts and the number of OpenOCD messages, its still fails.

OpenOCD log:
--------------
[20200902 10h00 console log.pdf](https://github.com/espressif/idf-eclipse-plugin/files/5160918/20200902.10h00.console.log.pdf)

Eclipse error message:
-----------------------
[image]

Error log:
---------
[image]

Error detail:
-----------
[image]

kolipakakondal, 2020-09-02:
Thanks for providing the logs @ammaree 

Here is the problem which still need to be fixed:
1. Go to startup Tab
2. Uncheck `Load executable` as mentioned in the screenshot.(We will document this)
[image]

kolipakakondal, 2020-09-02:
Particualr OpenOCD versoins have the following error.

```
Error: cpu0: Failed to write breakpoint instruction (-4)!
Error: cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Error: cpu0: Failed to write breakpoint instruction (-4)!
Error: cpu0: Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
```

If you have the above issue, please get the latest OpenOCD version and configure it the `Debugger` tab.
Download latest OpenOCD https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20200709

[image]

curiousmuch, 2020-09-02:
> Thanks @curiousmuch for providing the logs.
>
> Could you try this local build and confirm it. Based on that, I'll provide release update. [...]

I'll give this a shot tonight and let you know. Thanks again for the quick response!

curiousmuch, 2020-09-03:
@kolipakakondal this will work if the startup configuration is modified as suggested above (i.e. disable JTAG load).

closed 2020-09-03 by curiousmuch

kolipakakondal, 2020-09-03:
@curiousmuch Thanks for verifying this. We will push the update site soon with these changes.

curiousmuch, 2020-09-03:
Sounds good. Thanks for the quick responses! Do you know if the plug in will support JTAG load in the future? 

Just curious. I can get around it with a custom script.