espressif/idf-eclipse-plugin #119: Can not build project. No toolchain found for Target / Build not configured correctly

kifantidis, 2020-09-08:
**Describe the bug**
When I press build button I get the error message "No Toolchain found for Target esp32" or "Build not configured correctly". Also Launch Configurations doesn't open when pressing on the gear icon.

**To Reproduce**
Create new project from example hello world. Create ESP32 target. Press build button.

**Expected behavior**
Compiling the project.

**Screenshots**
[image]

[image]

**Environment Information:** 
Attached File: Enviroment Information.txt

OS: Windows 10 64bit
Python 3.8.5
Git 2.28.0
Notepad++ 7.8.8
Java SE 11 (jdk-11.0.8_windows-x64)
Eclipse 2020-06

Added to system PATH:
Git path
Python path
Java path

Idf path directory: C:/esp/esp-idf
Eclipse WorkSpace directory: C:/esp/eclipseWS

**ESP-IDF Tools Information:**
Executing C:\01_progs\Python\Python38-32\python.exe C:\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r2-8.2.0 (recommended)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r2-8.2.0 (recommended)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended)
* cmake: CMake build system
  - 3.13.4 (recommended)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended)
* ninja: Ninja build system
  - 1.9.0 (recommended)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended)

**IDF Eclipse Plugin Version:** 
ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD || ESP-IDF Eclipse Plugin || 1.2.2.202009041040 || 	com.espressif.idf.feature ||

kolipakakondal, 2020-09-09:
Hi @kifantidis 

Looks like CDT is unable to find the esp32 toolchain to build the project.

To verify whether the toolchains are identified correctly or not, please check the core build toolchain preferences Open 

1. Eclipse Preferences
2. Navigate to C/C++ -> Core Build Toolchains preference page

Please attach the screenshot for the same.

To reset the configuration, please do the following. 
1. Go to Eclipse preferences
2. C/C++ Build > Environment
3. Delete all variables configured here
4. Run this: Help > ESP-IDF Tools Manager > Install tools (this will reconfigure environment variables)
5. Run the project again by selecting the esp target

kifantidis, 2020-09-09:
Hello @kolipakakondal . Thank you for your response.

I've just tried all of your suggestions but I get the same behavior. I'm attaching some screenshots bellow.

1. Eclipse_C-C++_CoreBuildToolchains
[image]

2. Eclipse_C-C++_Build_Enviroment
[image]

3. Espressif IDF Tools Console After Install Tools.txt
[Espressif IDF Tools Console After Install Tools.txt](https://github.com/espressif/idf-eclipse-plugin/files/5193446/Espressif.IDF.Tools.Console.After.Install.Tools.txt)

4. Same_Behavior_Build_not_configured_correctly
[image]

kolipakakondal, 2020-09-09:
Hi @kifantidis 

This seems to an issue with installing the esp-idf tools. Here the message from the install tools log.

```
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\user_001\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\user_001\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
WARNING: 	unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\user_001\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
ERROR: Failed to download, and retry count has expired
```

Could you please check the following links and see if it helps.
https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/

https://github.com/espressif/esp-idf/issues/4081

kifantidis, 2020-09-09:
Hello @kolipakakondal .Thank you for helping me solve this issue. 
The problem was that python can't take the certification files from OS/browsers. In order to solve this I've installed certifi with pip in python and then I've run the command "python -m certifi" from command line. 
If anyone has the same problem as I did, use this link [How to install and use certifi module with pip](https://pypi.org/project/certifi/) .

After doing this you have to :

> To reset the configuration, please do the following.
>
>     1. Go to Eclipse preferences
>
>     2. C/C++ Build > Environment
>
>     3. Delete all variables configured here
>
>     4. Run this: Help > ESP-IDF [...]

And everything shall be fine.

Best regards, Kostas.

closed 2020-09-09 by kifantidis