espressif/idf-eclipse-plugin #121: Installation instructions followed, Java 11 installed, error installing idf-eclipse-plugin

mtnbarn, 2020-09-08:
Get the following error when installing idf-eclipse-plugin

Cannot satisfy dependency:
    From: C/C++ Autotools support 9.11.1.202004012023 (org.eclipse.cdt.autotools.feature.group 9.11.1.202004012023)
    To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.100.202001140305,1.1.100.202001140305]
  Cannot satisfy dependency:
    From: Template Engine Freemarker Support (Java 11) 1.1.100.202001140305 (org.eclipse.tools.templates.freemarker.java11 1.1.100.202001140305)
    To: osgi.ee; (&(osgi.ee=JavaSE)(version=11))

A clear and concise description of what the bug is.

**To Reproduce**
Install Java 11 LTS  "jdk-11.0.8_windows-x64_bin.exe"
check Python Version 3.8.2
Install eclipse - "eclipse-inst-win64.exe" select cpp option
Install idf-eclipse-plugin as per instructions - get error as above

**Expected behavior**
I expected this to work

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

[EclipseConfig.txt](https://github.com/espressif/idf-eclipse-plugin/files/5191173/EclipseConfig.txt)
**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

kolipakakondal, 2020-09-09:
Hi @mtnbarn 

It seems to be your eclipse is using Java 1.8.0_261 and that's causing the error reported.

I could see this from the log:
> C:\Program Files\Java\jre1.8.0_261

To verify, go to terminal and type `java`
Please set the JAVA_HOME correctly with the Java 11 and restart the eclipse to fix the problem

mtnbarn, 2020-09-09:
Thank you so much for your help

Before and after setting JAVA_HOME env. var. (Below cut from command
prompts)
*JAVA_HOME=C:\Program Files\Java\jdk-11.0.8*

java --version output was the same,

*C:\Users\owner>java --versionjava 11.0.8 2020-07-14 LTSJava(TM) SE Runtime
Environment 18.9 (build 11.0.8+10-LTS)Java HotSpot(TM) 64-Bit Server VM
18.9 (build 11.0.8+10-LTS, mixed mode)*

*Even restarted computer, still have following in eclipse configuration*

C:\Program Files\Java\jre1.8.0_261\bin\server\jvm.dll

During the install I put in the path to the Java 11 directory.

And further suggestions?

mtnbarn, 2020-09-09:
There does not seem to be any way to change this

[image: image.png]

kolipakakondal, 2020-09-09:
Okay, you just to need -vm argument to the eclipse.ini to force the eclipse to use Java version.

```
-vm
C:\Java\JDK\11.0\bin\javaw.exe
```

Please follow the instructions from here https://wiki.eclipse.org/Eclipse.ini

JAICHANGPARK, 2020-09-09:
I Fixed about JDK version 14. It's works to me ~

mtnbarn, 2020-09-09:
Thank you, that worked.

kolipakakondal, 2020-09-10:
Good to hear from you guys. Closing the issue as the issue got fixed.