espressif/idf-eclipse-plugin #125: Application Size Analysis crash (NullPointerException) eclipse plugin

shachar-limor, 2020-09-14:
Running the App Size Analysis crashes on Win10 64bit.

**To Reproduce**
Just run it on the blink example.
I used a single core but same results when came back to dual core (I am using DevKitC with SOLO  board)
Tried with both board attached (via USB) and without.

1. Build the blink target.
2. Right click project, then "Application Size Analysis crash"
3. Crash with error "Failed to create the part's controls" (see log below)

java.lang.NullPointerException
	at com.espressif.idf.ui.size.IDFSizeOverviewComposite.createPartControl(IDFSizeOverviewComposite.java:58)
	at com.espressif.idf.ui.size.IDFSizeAnalysisEditor.createOverviewPage(IDFSizeAnalysisEditor.java:114)
	at com.espressif.idf.ui.size.IDFSizeAnalysisEditor.createPages(IDFSizeAnalysisEditor.java:41)
	at org.eclipse.ui.part.MultiPageEditorPart.createPartControl(MultiPageEditorPart.java:333)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPartControl(CompatibilityPart.java:154)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityEditor.createPartControl(CompatibilityEditor.java:96)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:361)
	at jdk.internal.reflect.GeneratedMethodAccessor64.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:999)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:964)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalInject(InjectorImpl.java:140)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:405)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:332)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:202)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:91)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:60)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:42)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:132)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:1000)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:660)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:766)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:737)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:731)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:715)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl$1.handleEvent(PartServiceImpl.java:107)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler.lambda$0(UIEventHandler.java:38)
	at org.eclipse.swt.widgets.Synchronizer.syncExec(Synchronizer.java:236)
	at org.eclipse.ui.internal.UISynchronizer.syncExec(UISynchronizer.java:133)
	at org.eclipse.swt.widgets.Display.syncExec(Display.java:4735)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application$1.syncExec(E4Application.java:219)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler.handleEvent(UIEventHandler.java:38)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:205)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:203)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:234)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:151)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:132)
	at org.eclipse.equinox.internal.event.EventAdminImpl.sendEvent(EventAdminImpl.java:75)
	at org.eclipse.equinox.internal.event.EventComponent.sendEvent(EventComponent.java:44)
	at org.eclipse.e4.ui.services.internal.events.EventBroker.send(EventBroker.java:55)
	at org.eclipse.e4.ui.internal.workbench.UIEventPublisher.notifyChanged(UIEventPublisher.java:63)
	at org.eclipse.emf.common.notify.impl.BasicNotifierImpl.eNotify(BasicNotifierImpl.java:424)
	at org.eclipse.e4.ui.model.application.ui.impl.ElementContainerImpl.setSelectedElementGen(ElementContainerImpl.java:170)
	at org.eclipse.e4.ui.model.application.ui.impl.ElementContainerImpl.setSelectedElement(ElementContainerImpl.java:188)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.showElementInWindow(ModelServiceImpl.java:658)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.bringToTop(ModelServiceImpl.java:622)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.delegateBringToTop(PartServiceImpl.java:790)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.bringToTop(PartServiceImpl.java:404)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.showPart(PartServiceImpl.java:1266)
	at org.eclipse.ui.internal.WorkbenchPage.busyOpenEditor(WorkbenchPage.java:3195)
	at org.eclipse.ui.internal.WorkbenchPage.lambda$9(WorkbenchPage.java:3100)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:72)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3098)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3068)
[... 913 lines cut ...]
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

**IDF Eclipse Plugin Version:** 
ESP-IDF Eclipse Plugin

Version: 1.2.2.202009041040

Copyright (c) 2019 ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD. All rights reserved.

kolipakakondal, 2020-09-21:
Hi @shachar-limor 

This seems to be working only from idf release 4.3 and higher or master for now. We will fix this soon.

shachar-limor, 2020-09-22:
Hi Kondal,
Thanks for your comments.
As a nub in esp, I used the process described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html which got me the more "stable" versions installed (running "idf.py --version" shows me "ESP-IDF v4.1-dirty", this is from a few weeks ago.

So what is the recommended process to upgrade to 4.3? Is this an official stable release at this point or do I have to do a more manual process?

BTW, running the "install.bat" unmodified did not get me any further in updating the toolchain.

Regards,
Shachar

kolipakakondal, 2021-06-04:
This is fixed with v2.1.1 release, please get the latest changes using the update site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2021-06-04 by kolipakakondal