espressif/idf-eclipse-plugin #126: unresolved libraries

Kamalmohamed1, 2020-09-14:
i have just downloaded the plugin with a new eclipse and just trying a sample project to make sure all is fine and then this .
[image]

kolipakakondal, 2020-09-21:
Hi @Kamalmohamed1 

Project errors will be resolved only after the build.  Can you please check this https://github.com/espressif/idf-eclipse-plugin#BuildApplication.

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release. Recently we have released 2.0.0-beta as well, this is fixes for unresolved headers and inclusion issues, please get the latest changes using https://dl.espressif.com/dl/idf-eclipse-plugin/updates/beta/ update site

closed 2021-02-16 by kolipakakondal