espressif/idf-eclipse-plugin #127: Build Failure in Eclipse 2020-09 due to Removal of Deprecated Class

alemariusnexus, 2020-09-16:
I just updated Eclipse to version 2020-09 and reinstalled the ESP-IDF plugin (version 1.2.2). When building a project (imported using "Import -> Existing IDF Project"), I get the following error right at the start and the build stops:

```
java.lang.NoClassDefFoundError: org/eclipse/cdt/utils/Platform
	at com.espressif.idf.core.build.IDFBuildConfiguration.findCommand(IDFBuildConfiguration.java:588)
	at org.eclipse.cdt.core.build.CBuildConfiguration.startBuildProcess(CBuildConfiguration.java:493)
	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:252)
	at org.eclipse.cdt.core.build.CBuilder.build(CBuilder.java:45)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:832)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:220)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:263)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:316)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:319)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:278)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:467)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:565)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:603)
	at org.eclipse.core.internal.resources.Project.build(Project.java:116)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.lambda$0(LaunchConfigurationDelegate.java:406)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2312)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildProjects(LaunchConfigurationDelegate.java:412)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildForLaunch(LaunchConfigurationDelegate.java:122)
	at org.eclipse.launchbar.core.target.launch.LaunchConfigurationTargetedDelegate.superBuildForLaunch(LaunchConfigurationTargetedDelegate.java:61)
	at com.espressif.idf.launch.serial.internal.SerialFlashLaunchConfigDelegate.buildForLaunch(SerialFlashLaunchConfigDelegate.java:180)
	at org.eclipse.launchbar.ui.internal.commands.BuildActiveCommandHandler$1$1.run(BuildActiveCommandHandler.java:110)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.lang.ClassNotFoundException: org.eclipse.cdt.utils.Platform cannot be found by com.espressif.idf.core_1.0.0.202008190353
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:516)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:171)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	... 26 more
```

A bit of googling lead me to [this page](https://help.eclipse.org/2020-09/topic/org.eclipse.cdt.doc.isv/guide/deprecated_API_removals.html#cdtutilsPlatform) that states:

> Class org.eclipse.cdt.utils.Platform has been removed. Use org.eclipse.core.runtime.Platform instead.

Seems like that was a change from 2020-06 to 2020-09, and needs to be fixed in the ESP-IDF plugin in order to support Eclipse >= 2020-09. For the time being, I'm staying with 2020-06 where everything seems to work fine.

kifantidis, 2020-09-17:
Thank you for letting us know.

kolipakakondal, 2020-09-21:
Hi All, 

We will provide the fix for this soon to support the Eclipse 2020-09 version. Please continue to use Eclipse 2020-06 until then.

alkosa, 2020-09-21:
Ok. Thanks.

kolipakakondal, 2020-09-24:
Hi All, Here is the local build to try out. This will fix a build failure with the Eclipse 2020-09. 

[com.espressif.idf.update-1.2.2-SNAPSHOT.zip](https://github.com/espressif/idf-eclipse-plugin/files/5273952/com.espressif.idf.update-1.2.2-SNAPSHOT.zip)

How to install this build? Check this https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

Let me know if it helps

melvinItude, 2020-09-24:
Doesn't work unfortunately.

`java.lang.NoClassDefFoundError: org/eclipse/cdt/utils/Platform
	at com.espressif.idf.core.build.IDFBuildConfiguration.isLocal(IDFBuildConfiguration.java:343)
	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:191)
	at org.eclipse.cdt.core.build.CBuilder.build(CBuilder.java:45)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:832)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:220)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:263)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:316)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:319)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:278)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:467)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:565)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:603)
	at org.eclipse.core.internal.resources.Project.build(Project.java:116)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.lambda$0(LaunchConfigurationDelegate.java:406)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2312)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildProjects(LaunchConfigurationDelegate.java:412)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildForLaunch(LaunchConfigurationDelegate.java:122)
	at org.eclipse.launchbar.core.target.launch.LaunchConfigurationTargetedDelegate.superBuildForLaunch(LaunchConfigurationTargetedDelegate.java:61)
	at com.espressif.idf.launch.serial.internal.SerialFlashLaunchConfigDelegate.buildForLaunch(SerialFlashLaunchConfigDelegate.java:180)
	at org.eclipse.launchbar.ui.internal.commands.BuildActiveCommandHandler$1$1.run(BuildActiveCommandHandler.java:110)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.lang.ClassNotFoundException: org.eclipse.cdt.utils.Platform cannot be found by com.espressif.idf.core_1.0.0.202009221614
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:516)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:171)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	... 25 more
`

kolipakakondal, 2020-09-25:
@melvinItude Please check this https://github.com/espressif/idf-eclipse-plugin/pull/132/checks?check_run_id=1165362632

This should work. Please confirm

Lisa999, 2020-09-25:
> @melvinItude Please check this https://github.com/espressif/idf-eclipse-plugin/suites/1244487907/artifacts/18870276
> 
> This should work. Please confirm

Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'esp-idf-test'.
[Lorg/eclipse/cdt/cmake/core/internal/CompileCommand;

nikosamus, 2020-09-29:
Hi @kolipakakondal, I found this thread as I had the same problem. I've installed the last version and the build process moved forward but get stucked in a further step:

`!ENTRY org.eclipse.core.resources 4 566 2020-09-29 10:59:57.699
!MESSAGE Errors occurred during the build.
!SUBENTRY 1 org.eclipse.cdt.core 4 75 2020-09-29 10:59:57.699
!MESSAGE Errors running builder 'CDT Core Builder' on project 'yyyy'.
!STACK 0
java.lang.NoClassDefFoundError: [Lorg/eclipse/cdt/cmake/core/internal/CompileCommand;
	at com.espressif.idf.core.build.IDFBuildConfiguration.processCompileCommandsFile(IDFBuildConfiguration.java:359)
	at com.espressif.idf.core.build.IDFBuildConfiguration.build(IDFBuildConfiguration.java:323)
	at org.eclipse.cdt.core.build.CBuilder.build(CBuilder.java:45)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:832)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:220)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:263)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:316)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:319)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:278)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:467)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:565)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:603)
	at org.eclipse.core.internal.resources.Project.build(Project.java:116)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.lambda$0(LaunchConfigurationDelegate.java:406)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2292)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2312)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildProjects(LaunchConfigurationDelegate.java:412)
	at org.eclipse.debug.core.model.LaunchConfigurationDelegate.buildForLaunch(LaunchConfigurationDelegate.java:122)
	at org.eclipse.launchbar.core.target.launch.LaunchConfigurationTargetedDelegate.superBuildForLaunch(LaunchConfigurationTargetedDelegate.java:61)
	at com.espressif.idf.launch.serial.internal.SerialFlashLaunchConfigDelegate.buildForLaunch(SerialFlashLaunchConfigDelegate.java:180)
	at org.eclipse.launchbar.ui.internal.commands.BuildActiveCommandHandler$1$1.run(BuildActiveCommandHandler.java:110)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)`

Thanks

kolipakakondal, 2020-09-29:
Thanks for verifying. I"m able to reproduce. I'll update once the fix ready.

asidika, 2020-09-30:
Hi @kolipakakondal,  
I have tried previous version of IEC (V1.2.1) with  eclipse-2020-06 but unfortunately it does not work. 
Maybe, java version causes the problem but i m not sure.

kolipakakondal, 2020-10-01:
> Hi @kolipakakondal,
> I have tried previous version of IEC (V1.2.1) with eclipse-2020-06 but unfortunately it does not work.
> Maybe, java version causes the problem but i m not sure.

@asidika Could you check the Java version which is used? The plugin requires Java 11 and above.
https://github.com/espressif/idf-eclipse-plugin#Prerequisites

ainslec, 2020-10-01:
I have same bug on 2020-09, and using Java 12 (JDK).

asidika, 2020-10-01:
Hi @kolipakakondal my java version is JDK 15. Currently, i m trying to install from file IEC V1.2.1 with eclipse-2020-06. 
after restarting once plugin installed, i have faced pop-up below 
[image]

I have attached .log file

[log.log](https://github.com/espressif/idf-eclipse-plugin/files/5314755/log.log)

kolipakakondal, 2020-10-06:
> Hi @kolipakakondal my java version is JDK 15. Currently, i m trying to install from file IEC V1.2.1 with eclipse-2020-06.
> after restarting once plugin installed, i have faced pop-up below
> [image]
>
> I [...]

Looking at the log, your eclipse is started with the Java 1.8.0_261. Eclipse plugin requires Java 11 and above to work with. Could you please try changing the JAVA_HOME

kolipakakondal, 2020-10-06:
> Thanks for verifying. I"m able to reproduce. I'll update once the fix ready. Please continue to use **Eclipse 2020-06 (4.16)**

Hi All,

Eclipse 2020-09 support is added with this local build. This fix the issue we had with the OpenOCD Debugging launch. Could you please try and let me know how it goes. 
https://github.com/espressif/idf-eclipse-plugin/suites/1294855830/artifacts/20243567

To install the local build, please follow the instructions from here
https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

llemtt, 2020-10-06:
@kolipakakondal Eclipse 2020-09 now works, thanks!

andsoft1, 2020-10-08:
> Hi All,
>
> Eclipse 2020-09 support is added with this local build. This fix the issue we had with the OpenOCD Debugging launch. Could you please try and let me know how it goes. [...]

Hi,

I have encountered this same issue with eclipse 2020-09.
Did you solve the issue for idf eclipse plugin v.1.2.2?
I have tried to launch this set since 2 weeks but with no result.

Best regards
Andrzej

andsoft1, 2020-10-08:
This link 
https://github.com/espressif/idf-eclipse-plugin/suites/1294855830/artifacts/20243567
does not work.

BR
Andrzej

kolipakakondal, 2020-10-09:
Hi @andsoft1 Could you please attach the error log and details

rherrmannr, 2020-10-09:
It's working for me, too. Thanks!

jamawa-art, 2020-10-09:
Worked for me as well, thanks!

NHerreroReder, 2020-10-09:
> This link
> https://github.com/espressif/idf-eclipse-plugin/suites/1294855830/artifacts/20243567
> does not work.
> 
> BR
> Andrzej

Same problem for me. Cannot access to the file.

jamawa-art, 2020-10-09:
> > This link
> > https://github.com/espressif/idf-eclipse-plugin/suites/1294855830/artifacts/20243567
> > does not work.
> > BR
> > Andrzej
> 
> Same problem for me. Cannot access to the file.

I used it successfully only minutes ago. So maybe it is a firewall/privacy setting on your network? 
The link gave me a artifacts.zip which I used without unzipping for the "install software" step. This updated the plugin and solved the issue. 
So the link is just fine, 

I just realized that I am logged in at github, maybe that is the issue?

NHerreroReder, 2020-10-09:
> > This link
> > https://github.com/espressif/idf-eclipse-plugin/suites/1294855830/artifacts/20243567
> > does not work.
> > BR
> > Andrzej
> 
> Same problem for me. Cannot access to the file.

Finally I could access to the file. It works, thank you!! Using Eclipse 20-09 under Windows10, with JAVA 15,  GIT2.28, and esp-idf-v4.2-beta1
Two issues only:
  * I had to install some python requeriments (install.bat and export.bat inside esp-idf folder), and also had to manually install "gdbgui" with "python -m pip install pygdbmi==0.9.0.2 gdbgui==0.13.2.0".
  * Every time I restart Eclipse A popup window says "Load QML Analyzer" has encountered a problem. --> It seems related to Eclipse problems with JAVA 15 (suggested use JAVA 14 or wait for Eclipse update)

kolipakakondal, 2020-10-09:
Hi All,

ESP-IDF Eclipse Plugin 1.2.3 is released to support Eclipse 2020-09
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.3

Please update to the latest

andsoft1, 2020-10-09:
> 
> 
> Hi @andsoft1 Could you please attach the error log and details

Hi,
Thanks for support for all of you!
I was not logged on into the github.
Now I can build blink project with success ;-)

I see that v1.2.3 has arrived. So I am gonna test it.
Best regards
Andrzej