espressif/idf-eclipse-plugin #128: Issue with the include paths - Type 'esp_chip_info_t' could not be resolved

pavan-vibhute, 2020-09-20:
**Describe the bug**
I am running the hello world example project for ESP32 using Eclipse on Windows 10. Getting error as below. 

Error:
"Type 'esp_chip_info_t' could not be resolved". 

Version: esp-idf-v4.1

[image]
[image]

chechigp, 2020-09-21:
Same problem here :|

pavan-vibhute, 2020-09-21:
Simple hello world program should have run very smoothly. Looks like many people are facing same issue and yet there is no fix for this. 
Shall we switch to Linux GCC environment rather than eclipse for ESP32 ?

chechigp, 2020-09-21:
After expend a lot of hours trying to solve this problem with Eclipse without succees, I finally tried to move to Visual Studio Code, it took me an hour to work but finally I was able to compile and flash.
If anyone face the Python dependencies error in Visual Studio I solved it installing the minimum required version of the dependence (I don't know why but it give me a dependence error with higher versions).

I will keep an eye here if finally someone find the solution to this.

pavan-vibhute, 2020-12-03:
I reinstalled everything. I followed below sequence in installing. 
1. I followed the instructions on the below link and installed all perquisites. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html 
2. I got the ESP build system working using the ESP-IDF Command Prompt. I verified the build and flash works using the ESP-IDF command prompt. 
3. Then I followed the instructions in below link. 
 https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url 
4. The eclipse showed the issue where I had to enter the IDF_PYTHON_ENV_PATH manually. 
5. Now eclipse + ESP-IDF works for me.

kolipakakondal, 2021-02-16:
We have released 2.0.0-beta a, this has fixes for unresolved headers and inclusion issues, please get the latest changes using https://dl.espressif.com/dl/idf-eclipse-plugin/updates/beta/ update site

closed 2021-02-16 by kolipakakondal

ktwillia9, 2022-01-25:
For me, this worked so I didn't have to add the paths per project:
Under "Preferences > C/C++ > Build > Environment" select "Replace native environment with specified one"

also, python3 needs to be the default python. sudo apt install python-is-python3

ddomnik, 2022-05-12:
I am facing the same problem. The IDF has been just set up ... Selected Master when installing